jsalatas/plasma-pstate #1: Popup overflows desktop boundries on KDE Neon

arcooke, 2018-12-28:
Operating System: KDE neon 5.14
KDE Plasma Version: 5.14.4
Qt Version: 5.11.2
KDE Frameworks Version: 5.53.0
Kernel Version: 4.15.0-43-generic
OS Type: 64-bit
Processors: 8 × Intel® Core™ i7-6700K CPU @ 4.00GHz
Memory: 15.6 GiB of RAM

[image]

jsalatas, 2018-12-28:
Fixed https://github.com/jsalatas/plasma-pstate/commit/1a357ca1a47e55258593ad96e24b1537da0e0fff
Thanks!

arcooke, 2018-12-28:
I saw that commit come in before your reply and tested it and it didn't seem to help.  I just tested a 2nd time again and still no luck

I removed the widget from my panel, used `uninstall.sh`, re-cloned, and reinstalled again.  Is there something else I need to do to get this fix to kick in?

Verified my changes applied locally in `/usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/ui/main.qml` .. and the issue still exists

[image]

jsalatas, 2018-12-28:
Yeap. It can be very annoying while developing. :)

Happy that it worked!

closed 2018-12-28 by jsalatas