jsalatas/plasma-pstate #3: Dell Thermal mode is not showing up.

osleg, 2018-12-29:
I have Dell 9570 with smbios installed and loaded.
```
$ lsmod G smbios
dell_smbios            28672  2 dell_wmi,dell_laptop
dell_wmi_descriptor    16384  2 dell_wmi,dell_smbios
```

Also tried to check whether thermal settings are accessible 
```
sudo smbios-thermal-ctl -g
Helper function to Get current Thermal Mode settings

 Print Current Status of Thermal Information: 
-------------------------------------------------------------------
 
Current Thermal Modes: 
         Balanced
 
Current Active Acoustic Controller (AAC) Mode: 
         AAC mode Disabled
 
Current Active Acoustic Controller (AAC) Mode: 
        Global (AAC enable/disable applies to all supported USTT modes)
 
Current Fan Failure Mode: 
```

So it seems that thermal should work, yet widget doesn't show thermal settings.

jsalatas, 2018-12-29:
I have no idea why! lol :)
Just guessing: Is there a chance to be just scaling issue? Could you please the latest code? :\
If that guessing doesn't work, we will reconsider :)

osleg, 2018-12-29:
After the new version with fixed width I had that line for thermal settings for a moment but after opening pop up again it disappeared once again.
I tried to play with display scaling to see maybe some issues with it but nothing helps, the thermal part is just not there.

Is it possible to see any kind of logs for the plasmoid?

osleg, 2018-12-29:
[image]

here how it looks like btw

jsalatas, 2018-12-29:
I guess debug will not help much here. I believe there is some race conditions involved during the creation of the widget and it gets initialized before getting the reading.

I'll look at it. Thanks!

osleg, 2018-12-29:
A bit offtopic but, did you ever have different results with different thermal profiles? 
I just tested and it seems that it spins the fans at the very same speed with any thermal profile.

ontopic:
maybe it could be better to test for this at draw time instead of initialization time? I think plasmoids should be able to change content of popup dynamically? (Never wrote a plasmoid hence that's just a wild guess :) )

jsalatas, 2018-12-29:
> A bit offtopic but, did you ever have different results with different thermal profiles?
> I just tested and it seems that it spins the fans at the very same speed with any thermal profile.

It makes difference f your system is under load. I have performed some benchmarks using stress-ng that I may share at some time if I ever manage to process them an draw some conclusions. in general it seems that thermal profiles have some maximum fan speed limits which when reached it lowers the cpu speed.

jsalatas, 2018-12-29:
> maybe it could be better to test for this at draw time instead of initialization time?
lol! Actually just need to study the Qt documentation more instead of assuming things. :)

jsalatas, 2018-12-30:
Could you please try now after checking out the latest code? 

Hopefully it will work. If this is not the case, please reopen. 

Thanks again!

osleg, 2018-12-30:
Nope, still no luck ^_^

jsalatas, 2018-12-30:
lol! OK! Let's try to work it  out. What distro are you using? where is smbios-thermal-ctl installed?

osleg, 2018-12-31:
Arch linux
```
$ which smbios-thermal-ctl 
/usr/bin/smbios-thermal-ctl
```
Installed from official package
```
$ yay -Qi libsmbios
Name            : libsmbios
Version         : 2.4.2-1
Description     : A library for providing access to as much BIOS information as possible
Architecture    : x86_64
URL             : https://github.com/dell/libsmbios
Licenses        : GPL  custom
Groups          : None
Provides        : None
Depends On      : gcc-libs
Optional Deps   : python: tools [installed]
Required By     : fwupd  fwupdate
Optional For    : None
Conflicts With  : None
Replaces        : None
Installed Size  : 914.00 KiB
Packager        : Antonio Rojas <arojas@archlinux.org>
Build Date      : Fri 16 Nov 2018 05:49:29 AM +08
Install Date    : Mon 19 Nov 2018 11:07:53 AM +08
Install Reason  : Installed as a dependency for another package
Install Script  : No
Validated By    : Signature
```

jsalatas, 2018-12-31:
OK! Sorry about this but I guess we need to go step by step. 
could you please comment out the lines from 146 to 149 in set_prefs.sh script? 
https://github.com/jsalatas/plasma-pstate/blob/master/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh#L146

```
if check_dell_thermal; then
    json="${json},\"thermal_mode\":\"${thermal_mode}\""
fi

```
and replace these with the following?

`json="${json},\"thermal_mode\":\"cool-bottom\""`

that way we will know if it is bash/shell issue or a qml issue.

jsalatas, 2018-12-31:
Just a heads up: performing the above modifications  in manjaro when running in a virtual box seems to work. 

[image]

osleg, 2019-01-01:
Yup this change instantly fixed the issue

jsalatas, 2019-01-01:
So seems like it's the set_prefs.sh that doesn't send the correct information to qml  :\

Can we please try something more? Sorry about that :(

First of all undo the previous change. Then in `set_prefs.sh` do the following changes:
1) go to line 150
https://github.com/jsalatas/plasma-pstate/blob/master/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh#L150
and add before it the following

```
echo $json > /tmp/plasma_pstate.debug
```

2) go to line 146 
https://github.com/jsalatas/plasma-pstate/blob/master/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh#L146
and add before it the folowing

```
if check_dell_thermal; then
    echo "smbios-thermal-ctl found" > /tmp/check_dell_thermal.debug
else
    echo "smbios-thermal-ctl not found" > /tmp/check_dell_thermal.debug
fi
```

just wait a couple of seconds and look at the files `/tmp/check_dell_thermal.debug` and `/tmp/check_dell_thermal`.debug. What are their contents?

osleg, 2019-01-01:
```
$ cat plasma.out | jq
{
  "cpu_min_perf": "19",
  "cpu_max_perf": "100",
  "cpu_turbo": "false",
  "gpu_min_freq": "350",
  "gpu_max_freq": "1100",
  "gpu_min_limit": "350",
  "gpu_max_limit": "1100",
  "gpu_boost_freq": "1100",
  "gpu_cur_freq": "350",
  "cpu_governor": "powersave",
  "energy_perf": "balance_performance"
}
```

osleg, 2019-01-01:
```
$ cat /tmp/check_dell_thermal.debug 
smbios-thermal-ctl not found
```

osleg, 2019-01-01:
here is the culprit
I changed line 17 from  `smbios-thermal-ctl -g > /tmp/null 2>&1` to `smbios-thermal-ctl -g > /tmp/chk_out_res.out 2>&1`
```
$ cat chk_out_res.out 
Traceback (most recent call last):
  File "/usr/bin/smbios-thermal-ctl", line 37, in <module>
    locale.setlocale(locale.LC_ALL, '')
  File "/usr/lib/python3.7/locale.py", line 604, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting
```

jsalatas, 2019-01-01:
The error doesn't mean that it sees an empty LC_ALL value. It means that the locale in LC_ALL doesn't exist. You can verify that this is the case by running the following:

`sudo LC_ALL='' smbios-thermal-ctl -g`

and 

`sudo LC_ALL='xx_XX' smbios-thermal-ctl -g`

where xx_XX maybe whatever non-existent locale. The first command (with the empty LC_ALL) will work but the second will fail if the locale doesn't exist.

You can also try something like the following, which will generate a similar error. 

`LC_ALL='xx_XX' python -c "import locale; locale.setlocale(locale.LC_ALL, '')"`

however the following **should** work:

`LC_ALL='' python -c "import locale; locale.setlocale(locale.LC_ALL, '')"`

I could override the LC_ALL variable each time I call the `smbios-thermal-ctl` but I don't think that I really should try to fix broken/invalid locale settings :\

Please do a  search for 
````
  File "/usr/lib/pythonX.Y/locale.py", line nnn, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting
````
which seems to be common issue and I guess you will find the solution.

closed 2019-01-01 by jsalatas

osleg, 2019-01-06:
I see, I fixed it by just adding `LC_ALL=C ...` before each call to smbios in set_perfs script and now it works :)

osleg, 2019-01-06:
Thanks for assistance! <3