jsalatas/plasma-pstate #5: Widget keeps re-positioning itself in the top left corner of the desktop on every login.

arslancharyev31, 2019-01-03:
I have installed the latest version of the widget according to the instructions and it works as expected. However every time I login and start my session, I find it positioned at the top left corner of my desktop. Any suggestions how I can debug its behaviour in order to identify what causes such behaviour? Relevant system info: 
```
Operating System: Manjaro Linux 
KDE Plasma Version: 5.14.4
Qt Version: 5.12.0
KDE Frameworks Version: 5.53.0
Kernel Version: 4.19.12-2-MANJARO
```
P.S. I have not added user to the `sudoers` group as has been stated in the instructions, since the widget works for me without doing so.

jsalatas, 2019-01-03:
> However every time I login and start my session, I find it positioned at the top left corner of my desktop. 

I'm not sure what you mean here. Could you please provide a screenshot?

arslancharyev31, 2019-01-03:
> Could you please provide a screenshot?

Sure:
<details>
  <summary>Before restart:</summary>

[image]

</details>
<details>
  <summary>After Restart:</summary>
  
[image]

</details>

jsalatas, 2019-01-05:
Could you please try the latest commit?

arslancharyev31, 2019-01-06:
I did try it and appears to solver the problem, thank you.

P.S. I've noticed that widget's vertical dimension has increased a little bit, as compared to the previous version, but it's not that crucial.
<details>
  <summary>New version</summary>

[image]

</details>
<br>
P.S.S. It would be nice if the installer would auto-remove old version, since one has to manually remove `/usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate` folder before attempting to execute `install.sh`