jsalatas/plasma-pstate #6: Plans for Dedicated GPUS?

AdsonCicilioti, 2019-01-19:
Hi! Congratulation for this applet! Nice work.

Do you have plans for support Dedicated GPUs?

jsalatas, 2019-01-20:
Hi! 

Yes and no? :)

I guess my honest answer here is that I need to study a little bit both the nvidia/amdgpu drivers to see what options one could modify and what may be useful to have it in in you widget. 

Given the face that I personally never had the need to modify on the fly  nvidia/amd gpu parameters,  I would appreciate any suggestion on first of all what parameters/options one might be interesting to have in the widget.

steadfasterX, 2019-07-04:
> Given the fact that I personally never had the need to modify on the fly nvidia/amd gpu parameters, I would appreciate any suggestion on first of all what parameters/options one might be interesting to have in the widget.

First of all wow, AWESOME job dude.. really.. I had switched from GNOME recently and there I always had heavily used the [cpu power manager](https://extensions.gnome.org/extension/945/cpu-power-manager/) but missed always the possibility to control the GPU..

well I can speak for Nvidia only but here it would be great to have to possibility to change the 
"_Preferred PowerMizer mode_":

https://rastating.github.io/how-to-permanently-set-nvidia-powermizer-settings-in-ubuntu/

`nvidia-settings -q all | grep -C 10 -i powermizer`

```
  Attribute 'GPUCurrentPerfLevel' (masterbook:1[gpu:0]): 3.
    'GPUCurrentPerfLevel' is an integer attribute.
    'GPUCurrentPerfLevel' is a read-only attribute.
    'GPUCurrentPerfLevel' can use the following target types: X Screen, GPU.

  Attribute 'GPUAdaptiveClockState' (masterbook:1[gpu:0]): 1.
    'GPUAdaptiveClockState' is a boolean attribute; valid values are: 1 (on/true) and 0 (off/false).
    'GPUAdaptiveClockState' is a read-only attribute.
    'GPUAdaptiveClockState' can use the following target types: X Screen, GPU.

  Attribute 'GPUPowerMizerMode' (masterbook:1[gpu:0]): 2.
    Valid values for 'GPUPowerMizerMode' are: 0, 1 and 2.
    'GPUPowerMizerMode' can use the following target types: GPU.

  Attribute 'GPUPowerMizerDefaultMode' (masterbook:1[gpu:0]): 0.
    'GPUPowerMizerDefaultMode' is an integer attribute.
    'GPUPowerMizerDefaultMode' is a read-only attribute.
    'GPUPowerMizerDefaultMode' can use the following target types: GPU.

  Attribute 'ECCSupported' (masterbook:1[gpu:0]): 0.
    'ECCSupported' is a boolean attribute; valid values are: 1 (on/true) and 0 (off/false).
    'ECCSupported' is a read-only attribute.
    'ECCSupported' can use the following target types: GPU.

  Attribute 'ECCConfigurationSupported' (masterbook:1[gpu:0]): 0.
    'ECCConfigurationSupported' is a boolean attribute; valid values are: 1 (on/true) and 0 (off/false).
    'ECCConfigurationSupported' is a read-only attribute.
    'ECCConfigurationSupported' can use the following target types: X Screen, GPU.
```

So the main one is: `nvidia-settings -q GpuPowerMizerMode`:

```
  Attribute 'GPUPowerMizerMode' (masterbook:1[gpu:0]): 2.
    Valid values for 'GPUPowerMizerMode' are: 0, 1 and 2.
    'GPUPowerMizerMode' can use the following target types: GPU.
```

The following describes possible values (no sudo/root req):

- **Adaptive**: `nvidia-settings -a "[gpu:0]/GpuPowerMizerMode=0"`
- **Prefer max performance**: `nvidia-settings -a "[gpu:0]/GpuPowerMizerMode=1"`
- **Auto**: `nvidia-settings -a "[gpu:0]/GpuPowerMizerMode=2"`

Ofc there is more like `nvidia-settings -a GPUGraphicsClockOffset[3]=<some-negative-clock>` (untested) but I do not think it would be used a lot. 

The above though would be a very nice feature.. Let me know if you need anything else :)

jsalatas, 2019-07-05:
@steadfasterX thanks for your feedback and suggestion! I really appreciate it! 

please feel free to test the latest code in master branch.

steadfasterX, 2019-07-05:
> @steadfasterX thanks for your feedback and suggestion! I really appreciate it!
> 
> please feel free to test the latest code in master branch.

A.W.E.S.O.M.E !!!! works perfectly here :)
plasma 5.16 btw. 

2 little things: 

1. when set to Max Performance the dropdown is a bit to tiny.. could you adjust the width a bit?
2. when you are on it anyways: I would name the entry Power**M**izer so its like Nvidia spell it :)

[image]

Other then that thanks, really thats soo cool.. when the profiles are coming from my FR its 100% perfect :) 
( until I get new ideas, ofc xD )

jsalatas, 2019-07-05:
> when set to Max Performance the dropdown is a bit to tiny.. could you adjust the width a bit?
> when you are on it anyways: I would name the entry PowerMizer so its like Nvidia spell it :)

done

steadfasterX, 2019-07-05:
> > when set to Max Performance the dropdown is a bit to tiny.. could you adjust the width a bit?
> > when you are on it anyways: I would name the entry PowerMizer so its like Nvidia spell it :)
> 
> done

looks perfect now ....! :+1: thx again!