jsalatas/plasma-pstate #8: Support For LG laptop?

Sparsa, 2019-02-02:
Hi this is a very useful widget I wabted to add support for my LG Gram laptop, where i can add a switch to keep battery charge to 80%, and other changes. I went through your code and understood some part(new to qml). If you let me i will be able to add the sbove mentioned ideas. Thanks snd regards

jsalatas, 2019-02-02:
I would be more than happy to accept changes regarding the support of your laptop! 

I guess you can add a new section, like the thermal management which you may not see as this is available only to dell devices. 

In any case don't worry much about the gui as this is something that we can work together, but focus most reading/writing the relevant settings in [contents/code/set_prefs.sh](https://github.com/jsalatas/plasma-pstate/blob/master/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh ) and of course on properly detecting that the settings you want to change are available (see for example the check_dell_thermal functions and its uses ). 

Thank you so much!

jsalatas, 2019-02-10:
Could you please give it a try? If it works please feel free to close this issue.

Thanks again for your help!

Sparsa, 2019-02-12:
Hi, Thanks for this crucial job.
The battery charge limit is setting the limit to 80 but the checkbox remains unchecked and it can not be undone (though restart resets it)
USB Charge is always checked and can't be unchecked and clicking it doesn't change the value of the file
Fan Mode is working fine.

jsalatas, 2019-02-12:
could you please verify that the set_prefs.sh script works, ie test that every command below does what it is supposed to do? 

`sudo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh -lg-battery-charge-limit true`

`sudo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh -lg-battery-charge-limit false`

`sudo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh -lg-fan-mode true`

`sudo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh -lg-fan-mode false`

`sudo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh -lg-usb-charge true`

`sudo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh -lg-usb-charge false`

jsalatas, 2019-02-12:
> USB Charge is always checked and can't be unchecked and clicking it doesn't change the value of the file
Fan Mode is working fine.

Seems that usb charge and fan mode options work differently:

As per the [Documentation](https://github.com/torvalds/linux/blob/master/Documentation/laptops/lg-laptop.rst)

> Fan mode
> Writing **1/0** to /sys/devices/platform/lg-laptop/fan_mode **disables/enables** the fan silent mode.
> 
> USB charge
> Writing **0/1** to /sys/devices/platform/lg-laptop/usb_charge **disables/enables** charging another device from the USB port while the device is turned off.

Sparsa, 2019-02-13:
> could you please verify that the set_prefs.sh script works, ie test that every command below does what it is supposed to do?
>
> `sudo [...]

This all works fine :+1: 

> Seems that usb charge and fan mode options work differently:

Yes I know it, and I wrote the function this way, when it is checked(fan-mode) it means the silent mode is true and the value of the fan variable is 0 and when I uncheck it the silent mode goes off and the value of the file becomes 1. It is working perfectly.

Sparsa, 2019-02-13:
Another small request, is it possible to put the checkboxes battery limit and USB charge side by side instead of one on top of another? It will make the length of the widget a bit small and compact.

jsalatas, 2019-02-13:
> This all works fine +1

Great! I'll have a look. I guess (hope) that I just have some typos that need to be fixed.

> Another small request, is it possible to put the checkboxes battery limit and USB charge side by side instead of one on top of another? It will make the length of the widget a bit small and compact.

I'm in the process of redesigning it to a more compact size in order to fit in system tray (see #7). Although I personally like the current big touch friendly UI (I guess I'll offer the compact one as an alternative).

Sparsa, 2019-02-14:
> I'm in the process of redesigning it to a more compact size in order to fit in system tray (see #7). Although I personally like the current big touch friendly UI (I guess I'll offer the compact one as an alternative).
That's really awesome.  Actually, my screen is only 14'' so it looks really big when it expands so I asked you for that. Thanks for all your hard work again.
And I really want to learn QML scripts, can you point me where to learn? And also do you use some designer tool to design UI's or just write scripts by hand? It would be helpful if you let me know. 
Thanks again.

jsalatas, 2019-02-14:
I just tried to guess a fix. I'm confident that the "USB Charge" issue is fixed :\

jsalatas, 2019-02-14:
> I really want to learn QML scripts, can you point me where to learn? 

Hmmm...... I don't remember how I learned but I'm pretty sure that I just looked the existing qml code plasmoids :\
However I guess it's worth skimming through the [Qt5 Cadaques! Book](https://qmlbook.github.io/)

> also do you use some designer tool to design UI's or just write scripts by hand?

I usually use [Pencil](https://pencil.evolus.vn/) to experiment with the UI design and I'm writing the code in kate editor.

Sparsa, 2019-02-14:
Hi, I added a pull request which solves the minor bug in the shell script. Now everything is working fine.  Thank you for your support. I am closing this thread. :1st_place_medal:

closed 2019-02-14 by Sparsa