jsalatas/plasma-pstate #10: Empty widget screen

maloute, 2019-02-09:
Hello,

I have recently installed plasma-pstate on my Latitude 7490 running Kubuntu 18.10 using the install script.
I added the widget, but keeps getting a empty screen without any information:

[image]

I am quite new to the KDE environment (switched from xfce to kde recently), so perhaps i am missing something. What can I do to further diagnose this problem ?

Thank you

jsalatas, 2019-02-09:
Hmmm,..... that looks weird :\

what is the output of the following command? Does it ask you for your password when you run it?

`sudo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh`

maloute, 2019-02-09:
Thank you for your reply. It did not ask for my password, and here is the output (I just translated error codes):

> $ `sudo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh -read-all`
cat: /sys/devices/platform/lg-laptop/battery_care_limit: No file or directory found
cat: /sys/devices/platform/lg-laptop/usb_charge: No file or directory found
cat: /sys/devices/platform/lg-laptop/fan_mode: No file or directory found
{"cpu_min_perf":"9","cpu_max_perf":"100","cpu_turbo":"true","gpu_min_freq":"300","gpu_max_freq":"1150","gpu_min_limit":"300","gpu_max_limit":"1150","gpu_boost_freq":"1150","gpu_cur_freq":"300","cpu_governor":"powersave","energy_perf":"balance_performance","lg_battery_charge_limit":"","lg_usb_charge":"","lg_fan_mode":""} : command not found

jsalatas, 2019-02-09:
I assumed that you cloned/donloaded the master repository. Right? 
If that's the case, could you please try the latest release (1.0.3) from the link below? I believe that it will work for you. 

https://github.com/jsalatas/plasma-pstate/releases

maloute, 2019-02-09:
Yes Indeed I did that. I uninstalled and cloned the latest release and it works now, Thank you for your help !

closed 2019-02-09 by jsalatas

jordirelats, 2019-06-10:
I'm in Opensuse Tumbleweed and I have the same problem, If I run the following command
sudo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh-read-all
It ask the pasword and I get:
{"cpu_min_perf":"8","cpu_max_perf":"100","cpu_turbo":"true","gpu_min_freq":"300","gpu_max_freq":"1150","gpu_min_limit":"300","gpu_max_limit":"1150","gpu_boost_freq":"1150","gpu_cur_freq":"300","cpu_governor":"powersave","energy_perf":"balance_power","lg_battery_charge_limit":"false","lg_usb_charge":"false","lg_fan_mode":"true"}

What I need to do?

Thank you very much

jordirelats, 2019-06-10:
Sorry, I was on the SUDO group

Thanks Solved