jsalatas/plasma-pstate #14: Hiding unneeded entries "such as integrated GPU"

YamiYukiSenpai, 2019-02-24:
For those of us with dedicated GPUs, would it be possible to hide the GPU entry, or at least allow us to hide it?
------------------
Off-topic: the GitHub repo should include how to update the extension, just like in your KDE store page.

jsalatas, 2019-02-25:
Could you please verify and if everything is ok close the issue?

Thanks for you help!

YamiYukiSenpai, 2019-02-27:
My desktop has NVIDIA GPU.  I don't wanna call it a "waste of space", but seeing it there bothers me because it's disabled.
[image]

jsalatas, 2019-02-27:
Could you please checkout the latest code? I have added an option to hide it. I'm just asking you to confirm that this working for you. 
Thanks!

YamiYukiSenpai, 2019-02-27:
I assumed it wasn't released yet because it wasn't in the release section

I'll pull from master now

YamiYukiSenpai, 2019-02-27:
Didn't work

[image]

jsalatas, 2019-02-27:
Could you please remove the widget and then add it again?
BTW: there is a setting that you need to uncheck for this to work

YamiYukiSenpai, 2019-02-27:
I did do that
BTW, reboot fixed it
[image]

jsalatas, 2019-02-27:
OK. Thanis!

closed 2019-02-27 by jsalatas