jsalatas/plasma-pstate #15: install problem

IosebKoplatadze, 2019-03-01:
Error: Installation of /home/ioseb/snap/plasma-pstate/gr.ictpro.jsalatas.plasma.pstate failed: /home/ioseb/.local/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate already exists
chmod: cannot access '/usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh': No such file or directory

jsalatas, 2019-03-01:
What distro? 

How did you try to install it? By running the `install.sh` script as root (ie `sudo ./install.sh`) or something else?

IosebKoplatadze, 2019-03-02:
it's ubuntu 18.10
yes, I run 'sudo ./install.sh`

jsalatas, 2019-03-02:
can you
1) remove everything in /home/ioseb/.local/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate
sudo mv /home/ioseb/.local/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate ~/jsalatas_pstate
or (suggested)
sudo rm -r /home/ioseb/.local/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate
2) restart plasma or logout and then login again
3) run the sudo install thing again
4) restart plasma or logout and then login again (same as step 2)
and then try again? Please?
It should work!.... I hope! :)

IosebKoplatadze, 2019-03-02:
Thanks for the attention
I did your suggestions (step 1, step 2 ), but on step 3 there were new error message when I run 'sudo ./install.sh'

`ioseb@ioseb-XPS-13-9370:~/Downloads/plasma-pstate-1.0.4$ sudo ./install.sh 
Face detection timeout reached
[sudo] password for ioseb: 
Successfully installed /home/ioseb/Downloads/plasma-pstate-1.0.4/gr.ictpro.jsalatas.plasma.pstate
chmod: cannot access '/usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh': No such file or directory

Setup complete.
`
looks like it tries to find '/usr/share/plasma/plasmoids' but  
`ioseb@ioseb-XPS-13-9370:/usr/share/plasma$ ls

**desktoptheme**  **services**
`
there are only 'desktoptheme'  and 'services' folders

do I need some pre dependencies?

I'm sorry if I asking  questions like that, but I'm new with linux

jsalatas, 2019-03-02:
Is your distro ubuntu or kubuntu? Are you using kde? if not, I guess you will not be able to use it. :\

closed 2019-04-29 by jsalatas