jsalatas/plasma-pstate #17: Plasmoid window empty when clicked but shows info when hovering over icon

richterlevania3, 2019-03-25:
[image]

jsalatas, 2019-03-25:
What happens when you run the following command? Does it ask you for a password? 

`sudo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh -read-all
`

richterlevania3, 2019-03-25:
sudo doesn't work for me
I must su into root first, then the following shows up.

[image]

jsalatas, 2019-03-25:
The application requires you to be able to execute scripts via sudo. You should add yourself to the sudoers file.

closed 2019-05-17 by jsalatas