jsalatas/plasma-pstate #18: add support for amd cpu,apu and gpu.

librewish, 2019-04-24:


jsalatas, 2019-04-24:
Unfortunately I don't own any AMD hardware, so I can't really support it. However I'll accept any pull requests for such support, even the widget is named "Intel P-state.... " :)

Regarding NVidia support, as I already mentioned at https://github.com/jsalatas/plasma-pstate/issues/6  I personally never had the need to modify nvidia gpu parameters, and I would appreciate any suggestions about what options would you be interesting in modifying through the widget.

cmpute, 2019-11-21:
I have AMD processors and wish to help, but have no idea how to contribute...

cmpute, 2020-01-06:
find this repo to be useful: https://github.com/kevinlekiller/amdctl
I have already set it up and I'm trying to test if it can set pstate of my AMD Ryzen 2700

other repos that might be useful: https://github.com/ocerman/zenmonitor

cmpute, 2020-01-06:
I can get the p-state information from this tool by `sudo ./amdctl -g -c0`
```
Voltage ID encodings: SVI (serial)
Detected CPU model 8h, from family 17h with 16 CPU cores.

Core 0 | P-State Limits (non-turbo): Highest: 3 ; Lowest 3 | Current P-State: 3
 Pstate Status CpuFid CpuDid CpuVid CpuMult CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
      0      1    128      8     88  32.00x 3200MHz  1000mV     32     10  42.00A   42.00W
      1      1    140     10    108  28.00x 2800MHz   875mV     28     10  38.00A   33.25W
      2      1    124     16    122  15.50x 1400MHz   788mV     15     10  25.00A   19.69W
      3      disabled
Northbridge:
No P-States on AMD17H Northbridge.
```
But I actually have no idea how can I set the p-state, this tool enables modification of voltages and frequencies of p-states

jsalatas, 2020-01-06:
> I have already set it up and I'm trying to test if it can set pstate of my AMD Ryzen 2700

That might be interesting! I'm a little bit busy at that time (and I guess for at least a couple of months more) but in any case, if you could create a PR, I'll for surre merge it..... at some time :\

Thanks for your help and feedback!

kermeat, 2020-01-30:
Take a look at cpupower. It can handle all supported by kernel cpu
https://github.com/torvalds/linux/tree/master/tools/power/cpupower

lapineige, 2020-09-17:
Ryzen 5 3600 here, I'm willing to help to, but I can't contribute do any code, sorry. AMD support would be very neat :)
I'm ready to test something if you wish. 
(I also have an intel CPU on another computer, if you need to test something on both)
I'm my case I'm specially interested in the CPU governor parameter, to have a one-click tool to switch between power saving, "balanced" and full performance. Overclocking/changing frequencies could be a nice addition, but it's not important for me.

jsalatas, 2020-09-17:
> Ryzen 5 3600 here, I'm willing to help to, but I can't contribute do any code, sorry. AMD support would be very neat :)
> I'm ready to test something if you wish.

I would appreciate your feedback in a couple of questions. I'm actually implementing part of my widget in the upstream KDE, and there are some open questions regarding AMD support. 

Could you please email me and I'll come back to you. 

Thanks for your help!

librewish, 2020-09-17:
Hope this helps

I just added some extra things to your scripts

 performance.sh
`
#!/bin/sh
    gov=performance
    if [ -n "$gov" ]; then
        for cpu in /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor; do
            printf '%s\n' "$gov" > $cpu; 2> /dev/null
        done
    fi
    energyperf=performance
    if [ -n "$energyperf" ]; then
        if [ -f /sys/devices/system/cpu/cpu0/cpufreq/energy_performance_preference ]; then
            for cpu in /sys/devices/system/cpu/cpu*/cpufreq/energy_performance_preference; do
                printf '%s\n' "$energyperf" > $cpu; 2> /dev/null
            done
        else
            pnum=$(echo $energyperf | sed -r 's/^performance$/0/;
                                s/^balance_performance$/4/;
                                s/^(default|normal)$/6/;
                                s/^balance_power?$/8/;
                                s/^power(save)?$/15/')
            x86_energy_perf_policy $pnum > /dev/null 2>&1
        fi
    fi
    nvidia-settings -a "[gpu:0]/GpuPowerMizerMode=1" 2> /dev/null
    echo performance > /sys/class/drm/card0/device/power_dpm_state
    echo high > /sys/class/drm/card0/device/power_dpm_force_performance_level
    echo performance > /sys/module/pcie_aspm/parameters/policy
    echo 0 > /sys/devices/system/cpu/intel_pstate/no_turbo
    echo 1 > /sys/devices/system/cpu/cpufreq/boost
`

Powersave.sh

`#!/bin/sh

    gov=powersave
    if [ -n "$gov" ]; then
        for cpu in /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor; do
            printf '%s\n' "$gov" > $cpu; 2> /dev/null
        done
    fi

    energyperf=power
    if [ -n "$energyperf" ]; then
        if [ -f /sys/devices/system/cpu/cpu0/cpufreq/energy_performance_preference ]; then
            for cpu in /sys/devices/system/cpu/cpu*/cpufreq/energy_performance_preference; do
                printf '%s\n' "$energyperf" > $cpu; 2> /dev/null
            done
        else
            pnum=$(echo $energyperf | sed -r 's/^performance$/0/;
                                s/^balance_performance$/4/;
                                s/^(default|normal)$/6/;
                                s/^balance_power?$/8/;
                                s/^power(save)?$/15/')

            x86_energy_perf_policy $pnum > /dev/null 2>&1
        fi
    fi

    nvidia-settings -a "[gpu:0]/GpuPowerMizerMode=2" 2> /dev/null
    echo battery > /sys/class/drm/card0/device/power_dpm_state
    echo low > /sys/class/drm/card0/device/power_dpm_force_performance_level
    echo powersave > /sys/module/pcie_aspm/parameters/policy
    echo 1 > /sys/devices/system/cpu/intel_pstate/no_turbo
    echo 0 > /sys/devices/system/cpu/cpufreq/boost
`

lapineige, 2021-06-12:
>  I'm actually implementing part of my widget in the upstream KDE

Hello @jsalatas, do you have any piece of new regarding that ?

jsalatas, 2021-06-12:
Hi! This is handled by Kai Uwe (from the KDE team) and hopefully we will see some initial implementation in 5.23 

https://invent.kde.org/plasma/powerdevil/-/merge_requests/34#note_230681

Notice that recent kernels have introduced the platform profile API (https://www.kernel.org/doc/html/latest/userspace-api/sysfs-platform_profile.html) and I'm not sure at the moment how many devices and configurations are supported by that API.