jsalatas/plasma-pstate #23: batery limit don't work on opensuse Tumbleweed and LG Gram 17

jordirelats, 2019-06-10:
On opensuse tumbleweed and LG Gram 17" the battery limit can not be setup, when I select after few seconds it get unselect?

jsalatas, 2019-06-10:
Hmmm.... no idea! :(

does it change if you echo directly to the related sysfs entry? 
does it change if you use directly the set_prefs.sh script from the command line?

Maybe @Sparsa could help here? :\

Sparsa, 2019-06-11:
Hi, can you check which kernel are you using? Also if that kernel contains the driver module or not.

jordirelats, 2019-06-11:
> Hi, can you check which kernel are you using? Also if that kernel contains the driver module or not.
I'm using Plasma  5.15.5, Frameworks 5.58.0, qt 5.12.3, Kernel 5.1.7-1.
The Kernel I thing it contains the driver module because it works the special keys like keyboard back ligth

jordirelats, 2019-06-11:
> Hmmm.... no idea! :(
> 
> does it change if you echo directly to the related sysfs entry?
> does it change if you use directly the set_prefs.sh script from the command line?
> 
> Maybe @Sparsa could help here? :\

My linux know how is a litle limited, you can explain how I can make this test? Thank you very much

jsalatas, 2019-06-11:
> My linux know how is a litle limited, you can explain how I can make this test? Thank you very much

could you please run the following commands and post the output?

```
sudo cat /sys/devices/platform/lg-laptop/battery_care_limit
sudo echo 80 > /sys/devices/platform/lg-laptop/battery_care_limit
sudo cat /sys/devices/platform/lg-laptop/battery_care_limit
sudo echo 100 > /sys/devices/platform/lg-laptop/battery_care_limit
sudo cat /sys/devices/platform/lg-laptop/battery_care_limit
sudo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh -lg-battery-charge-limit true
sudo cat /sys/devices/platform/lg-laptop/battery_care_limit
sudo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh -lg-battery-charge-limit false 
sudo cat /sys/devices/platform/lg-laptop/battery_care_limit

```
Thanks!

jordirelats, 2019-06-11:
> could you please run the following commands and post the output?
>
> ```
> sudo cat /sys/devices/platform/lg-laptop/battery_care_limit
> sudo echo 80 > /sys/devices/platform/lg-laptop/battery_care_limit [...]
This is the output
jordi@linux-psmc:~> sudo cat /sys/devices/platform/lg-laptop/battery_care_limit
0
jordi@linux-psmc:~> sudo echo 80 > /sys/devices/platform/lg-laptop/battery_care_limit
bash: /sys/devices/platform/lg-laptop/battery_care_limit: S’ha denegat el permís
jordi@linux-psmc:~> sudo cat /sys/devices/platform/lg-laptop/battery_care_limit
0
jordi@linux-psmc:~> sudo echo 100 > /sys/devices/platform/lg-laptop/battery_care_limit
bash: /sys/devices/platform/lg-laptop/battery_care_limit: S’ha denegat el permís
jordi@linux-psmc:~> sudo cat /sys/devices/platform/lg-laptop/battery_care_limit
0
jordi@linux-psmc:~> sudo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh -lg-battery-charge-limit true
jordi@linux-psmc:~> sudo cat /sys/devices/platform/lg-laptop/battery_care_limit
0
jordi@linux-psmc:~> sudo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh -lg-battery-charge-limit false
jordi@linux-psmc:~> sudo cat /sys/devices/platform/lg-laptop/battery_care_limit
0
Thanks

jsalatas, 2019-06-11:
"S’ha denegat el permís" aka "Permission denied"! 

Well some of the commands I gave you, are probably wrong! sorry about that! 

Anyways..... 0 should be a valid value for battery_care_limit. It should be either 80 or 100 I guess, according to the documentation :\

https://github.com/torvalds/linux/blob/v5.1/Documentation/laptops/lg-laptop.rst

Could you please test again the following? 

> sudo cat /sys/devices/platform/lg-laptop/battery_care_limit
> echo 80 | sudo tee /sys/devices/platform/lg-laptop/battery_care_limit
> sudo cat /sys/devices/platform/lg-laptop/battery_care_limit
> echo 100 | sudo tee /sys/devices/platform/lg-laptop/battery_care_limit
> sudo cat /sys/devices/platform/lg-laptop/battery_care_limit
> sudo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh -lg-battery-charge-limit true
> sudo cat /sys/devices/platform/lg-laptop/battery_care_limit
> sudo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh -lg-battery-charge-limit false 
> sudo cat /sys/devices/platform/lg-laptop/battery_care_limit
>

jordirelats, 2019-06-11:
"S’ha denegat el permís" aka "Permission denied"! Yes this is correct

This is the output
`jordi@linux-psmc:~> sudo cat /sys/devices/platform/lg-laptop/battery_care_limit
0
jordi@linux-psmc:~> echo 80 | sudo tee /sys/devices/platform/lg-laptop/battery_care_limit
80
jordi@linux-psmc:~> sudo cat /sys/devices/platform/lg-laptop/battery_care_limit
0
jordi@linux-psmc:~> echo 100 | sudo tee /sys/devices/platform/lg-laptop/battery_care_limit
100
jordi@linux-psmc:~> sudo cat /sys/devices/platform/lg-laptop/battery_care_limit
0
jordi@linux-psmc:~> sudo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh -lg-battery-charge-limit true
jordi@linux-psmc:~> sudo cat /sys/devices/platform/lg-laptop/battery_care_limit
0
jordi@linux-psmc:~> sudo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh -lg-battery-charge-limit false 
jordi@linux-psmc:~> sudo cat /sys/devices/platform/lg-laptop/battery_care_limit
0
jordi@linux-psmc:~> 
`

Thanks

jsalatas, 2019-06-11:
OK! Thanks! As I previously wrote 

> 0 should be a valid value for battery_care_limit. It should be either 80 or 100 I guess, according to the documentation :\
> 
> https://github.com/torvalds/linux/blob/v5.1/Documentation/laptops/lg-laptop.rst

Stupid question but I have to ask: Does your laptop support this feature? :)

jordirelats, 2019-06-11:
The Laptop is a LG Gram 17", for what I read it should support this feature.
May be I should configure better some thing, but I don't know.
I need to modify the files indicated in:[https://github.com/torvalds/linux/blob/master/Documentation/laptops/lg-laptop.rst](url)

Thank you

jsalatas, 2019-06-11:
> I need to modify the files indicated in:https://github.com/torvalds/linux/blob/master/Documentation/laptops/lg-laptop.rst

Obviously you can't modify that! :)

Unfortunately, I can't propose a solution/workaround for the moment, but I promise to try to investigate it and come back to you as soon as possible. Sorry! :(

jordirelats, 2019-06-11:
Thank you very much, sorry I'm not expert.

Sparsa, 2019-06-12:
> 
> 
> > Hi, can you check which kernel are you using? Also if that kernel contains the driver module or not.
> > I'm using Plasma  5.15.5, Frameworks 5.58.0, qt 5.12.3, Kernel 5.1.7-1.
> > The Kernel I think it contains the driver module because it works the special keys like keyboard back light

Actually, for keyboard special keys, the module is not needed. It was working without the module in my case. The module is needed when you want to have battery limit or speed control of fan etc. 
In my case, the fan speed control is not working anymore. Don't know why it happened maybe some firmware update broke it. But I can live with it. For your case what you can do is the following:
1. check if the driver is actually enabled ( "zgrep LG_LAPTOP /proc/config.gz", If it enabled then it should output "# /sys/devices/platform/lg-laptop")
2. If you find the files but not able to set the values then I am afraid that it might be some firmware changes. In that case, you can directly mail (matan[at]svgalib[dot]org) the author of the driver. I cant guarantee if he will reply or not, but you can try.
Regards,
Sparsa

jordirelats, 2019-06-12:
> 1. check if the driver is actually enabled ( "zgrep LG_LAPTOP /proc/config.gz", If it enabled then it should output "# /sys/devices/platform/lg-laptop")

I have checked and this are the result:
`jordi@linux-psmc:~> zgrep LG_LAPTOP /proc/config.gz

CONFIG_LG_LAPTOP=m
`What that means?

Thank you very much

Jordi

Sparsa, 2019-06-13:
> I have checked and this are the result:
> `jordi@linux-psmc:~> zgrep LG_LAPTOP /proc/config.gz
>
> CONFIG_LG_LAPTOP=m
> `What that means?
>
> Thank you very much
>
> Jordi

Hi,  CONFIG_LG_LAPTOP=m means that the driver is in module form. Which is correct. I was wrong in the previous comment. Anyway, as it seems that your kernel has the driver in module form. To check if it is loaded during boot do the following:
`❯ lsmod | grep lg_laptop
lg_laptop              20480  0
sparse_keymap          16384  1 lg_laptop
wmi                    32768  3 intel_wmi_thunderbolt,lg_laptop,wmi_bmof
`
and lg_laptop should be in the output as it is shown.

Regards,
Sparsa

jordirelats, 2019-06-13:
> Hi, CONFIG_LG_LAPTOP=m means that the driver is in module form. Which is correct. I was wrong in the previous comment. Anyway, as it seems that your kernel has the driver in module form. To check if it is loaded during boot do the following:
> `❯ lsmod | grep lg_laptop`

Hi

This is the out:
`jordi@linux-psmc:~> lsmod | grep lg_laptop

lg_laptop              20480  0
sparse_keymap          16384  1 lg_laptop
wmi                    32768  3 intel_wmi_thunderbolt,lg_laptop,wmi_bmof`

This is correct?

Regards

Jordi

Sparsa, 2019-06-13:
Yes, it is correct. Then I am sorry I have no idea why your laptop is not
supporting the driver. Please contact the author. Maybe he can do something.

Regards,
Sparsa

jsalatas, 2019-06-22:
@jordirelats just a wild guess: are you in the latest version of BIOS? 🤔

jordirelats, 2019-06-22:
> @jordirelats just a wild guess: are you in the latest version of BIOS?

My Laptop is 17Z990-V.AA75B
The Bios is W1ZD1180 X64

I have check and I thing is the latest Bios,

Regards
Jordi

jordirelats, 2019-09-18:
Hi,
Yesterday I realise that I have not the last Bios, I update the bios using Windows device manager, and automatic update the firmware. Now is sure that I have the last Bios. But still have the same problem.

I have entered to the hidden additional bios settings and I put 80% as limit of charge, but unfortunately the LG Gram charges until 100%.

I have the feeling that I need to change some parameters on the bios or is missing one package in order to make it work

jsalatas, 2019-09-23:
> I have the feeling that I need to change some parameters on the bios or is missing one package in order to make it work

Unfortunately it's almost impossible for me to look into that issue, as I don't own that hardware. I'm sorry about that :(

please keep us posted in case you find a solution. Thanks!

jordirelats, 2019-09-23:
Hi,
Finally the developer of the kernel module has answered my email, it seams that 2019 LG Gram have some issues with the module and He is working to solved. The problem is no your app, the problem it is in the kernel module, when it is solved I will post.

Thank you very much for your help

Sparsa, 2019-09-23:
I'm glad to know this. Excellent! will wait for the new update of the driver, I hope then my issue with silent fan mode will also be resolved :man_dancing: