jsalatas/plasma-pstate #24: Better fix for system log

principis, 2019-06-19:
I just saw #16 and I was using the workaround on the wiki for my Dell laptop but it doesn't fix all messages so I altered it a bit. I don't really know C which is why I didn't want to make a PR but if this fixes everything and doesn't create a security risk it would be nice to add it. :)

```
#include <unistd.h>

int main(int argc, char *argv[])
{
    setuid(0);
    execv("/usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh", argv);
    return 0;
}
```

jsalatas, 2019-06-20:
I guess you need also to change the command in the plasmoid to pass the -read-all parameter (the argv) right? Is this really a good idea to a setuid executable in a known path that accepts and (somehow) processes and uses command line arguments?

principis, 2019-06-22:
(For some reason I did not receive a notification...)
Yes, I changed it to pass `-read-all`. It should be fine or there could be a bug in set_pref.sh (or for example smbios-thermal-ctl) so that someone could write to arbitrary files. Every user can now change the thermal mode off course. I don't care but some users might. (I hadn't thought of that before :) )

principis, 2019-06-24:
I'll close this issue. I think it's a bad idea to recommend without warnings. Although in my opinion it's fine to use on your own system. An option would be to add it to the wiki with a clear warning for people like me who don't want any useless messages in syslog.

closed 2019-06-24 by principis