jsalatas/plasma-pstate #25: Wrong text padding on hiDPI display

zeronote, 2019-06-24:
Hi,

first of all, I have to say that I really appreciate your work, the plasma-pstate widget is extremely useful on my Dell xps.

One thing (that perhaps is not a real issue) I noticed is that on a hiDPI display some icon/text items are a little bit messed up, as you can see from the attached screenshot.

[image]

jsalatas, 2019-06-24:
I guess you need to adjust your small size font to a smaller value. In the screenshot below are the related settings from my desktop PC: as you can see I have a 2560x1440 resolution with no scaling and my small font size is 8 pt. In my laptop (XPS 15 with 4K screen) I have similar settings and it appears also OK. 

[image]

Could you please try reducing your small font size and let me know if that works for you and if you are OK with this?

PS: I'm glad you find the widget useful!

zeronote, 2019-06-24:
That was it! Setting my small font size to 8 pt fixed it.

I should say that I am on a fresh Arch + Plasma 5.16.1 installation and I'm struggling with
plenty of new regressions: the Font setting page can't read my actual DPI value (every time I set it I can see the changes but the next time I access the Font setting page DPI value is back to 96) and the default Small font size is 10 pt instead of 8 pt, that has been the default Small font value in KDE for long time.  

Thanks for the fast response, hope to help you with #7 soon.

closed 2019-06-24 by zeronote

jsalatas, 2019-07-11:
> Thanks for the fast response, hope to help you with #7 soon.

just a heads up here. Apparently for the support of the nvidia GPU I created a ComboBox element which should work with the same input as the Group ui element. So probably replacing [line 85 in Header.qml](https://github.com/jsalatas/plasma-pstate/blob/7de74d5c3951d4ce65b012e457a288062e81fc04/gr.ictpro.jsalatas.plasma.pstate/contents/ui/Header.qml#L85)
with the following 
````
combobox.createObject(controls, {'props': items[i]})
````
should be good for the starting point of a more compact ui (ComboBox instead the fancy Groups with PushButtons). After that it should be (hopefully) just a font properties tweaks.