jsalatas/plasma-pstate #29: Missing Bumblebee support

HeyBoy85, 2019-07-20:
If the computer is using bumblebee to manage the nvidia card, `nvidia-settings` will fail.
In order to use the `nvidia-settings` is needed to execute it through optirun utility (`optirun -b none nvidia-settings -c :8`, then you can add the other parameters you need).
I have already checked on my machine and:

- `optirun -b none nvidia-settings -c :8 -q GpuPowerMizerMode > /dev/null 2>&1` correctly terminate with `0` while `nvidia-settings -q GpuPowerMizerMode > /dev/null 2>&1` with 1.

- `optirun -b none nvidia-settings -c :8 nvidia-settings -c :8 -a "[gpu:0]/GpuPowerMizerMode=2"` correctly set the PowerMize mode.

- `optirun -b none nvidia-settings -c :8 -q GpuPowerMizerMode | grep "Attribute 'GPUPowerMizerMode'" | awk -F "): " '{print $2}'  | awk -F "." '{print $1}'` print the correct value.

The only think I'm not totally sure is how to check if Bumblebee is present and used, the easy, but probably not the correct way, is to check the return value of `command -v optirun`.

jsalatas, 2019-07-20:
Hmmm.... I'm not sure I would want to do it or how (I don't use bumblebee myself) but I for sure would accept any PR request if someone would implement it.

BTW: as I understand the `:8` parameter is the screen number. Right? Do you know if this is always the case, or if that might be some other arbitrary value?

linuskr, 2020-05-31:
According to [Bumblebee](https://github.com/Bumblebee-Project/Bumblebee/wiki/FAQ), it is always `:8`