jsalatas/plasma-pstate #30: Giant fonts on hover

alexgallotta, 2019-08-09:
When hovering, fonts are giants and I can't find any way to reduce them`
[image]
`

jsalatas, 2019-08-09:
Hmmmm.... the font-sizes are hardcoded according to what looked/looks OK in my PC :)

I'll have a second look though.... 

In the mean time, are you sure you don't have any issues with your fonts? It seems (please correct me if I'm wrong) that in general you are using a serif font, however the "C" letter (the celsius unit in temperature) seems like a non-serif (like it is substituted with some other font) :\

What distro are you using? 
What are you KDE font settings? Could you please also provide a screenshot of systemsettings -> fonts?

Thank you!!!

alexgallotta, 2019-08-09:
Here are my fonts
[image]
and here it is how it looks like when the icon opened
[image]
you can probably guess the difference in size between what's inside the widget and outside

jsalatas, 2019-08-09:
Oh! OK! I see what you mean! The difference isn't so apparent using the default fonts. :)

alexgallotta, 2019-08-09:
If this may help: reducing general font size seems to "reduce" the total size of the pop up
[image]

jsalatas, 2019-08-10:
could you please try now? :\

BTW: Apparently my symbol font sucks (it has the wrong metrics or something).

closed 2019-08-10 by jsalatas

alexgallotta, 2019-08-11:
I upgraded and restarted the machine, it did not work.
Anyway it's not a big deal if none else is affected

commit 7fbaa72b (2019-08-10): Giant fonts on hover

fixes #30