jsalatas/plasma-pstate #31: FR: set up values for battery and AC (TLP integration)

alexgallotta, 2019-08-09:
It would be great to have an integration with TLP.
In this way it would be easier to change settings in AC and battery mode
TLP offers tons of settings (from `tlp-stat`):
```
+++ Configured Settings: /etc/default/tlp
CPU_SCALING_GOVERNOR_ON_AC=performance
CPU_SCALING_GOVERNOR_ON_BAT=powersave
```
and info about where/what to customize
```
/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = performance
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors = performance powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   800000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  4000000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/energy_performance_preference = performance
/sys/devices/system/cpu/cpu0/cpufreq/energy_performance_available_preferences = default performance balance_performance balance_power power 
```

jsalatas, 2019-08-09:
Integrate it how? 
The settings you are mentioning (scaling_governor, scaling_min_freq, scaling_max_freq, energy_performance_preference) are already in the widget. 
What more do you need? A use case would be useful (ie: "I want to be able to do this and this and that through the widget")

alexgallotta, 2019-08-09:
Sorry I was confusing.
I meant only
> In this way it would be easier to change settings in AC and battery mode
so to give the chance to set up some on-battery settings and on-AC settings.

jsalatas, 2019-08-09:
Would CPU profiles (see issue #26) work for you?

alexgallotta, 2019-08-13:
That will be handy yes

alexgallotta, 2019-09-08:
I think this can be closed.
Configuring CPU for AC and non-AC mode from TLP would do the job, so it's probably counter productive to have the same feature here

closed 2019-09-08 by alexgallotta