jsalatas/plasma-pstate #35: Support MSI GS75 and others

ssergio-ll, 2019-09-23:
Hello.

Can you add support of control fan for MSI GS75 and others? I think that could you help with the project of isw (https://github.com/YoyPa/isw) and add to plamoid.

And could also add support to select Nvidia o Intel profile?

Thanks!

jsalatas, 2019-09-23:
> Can you add support of control fan for MSI GS75 and others?

I don't think I can do it as I don't own any MSI GS75 (and others) hardware :)

However I will gladly accept any PR requests regarding this issue. 

>  And could also add support to select Nvidia o Intel profile?

I'm not sure what you mean here. Could you please explain?

ssergio-ll, 2019-09-23:
Can you integrate the project https://github.com/YoyPa/isw to plasmoid widget. The laptop MSI have a option for fans call cooler boost trinity. This project can do run fans to max power (manually) from console, for example when you go to game. 

It would be interesting if you could add a option to plamoid with a checkbox that enable or disable this option. The functionality is made (https://github.com/YoyPa/isw). You just have to integrate it in plasmoid.

For example, now if i want active this, i have that go to console and write "sudo isw -b GS75_8SE on" and "sudo isw -b GS75_8SE off" for deactivate.

Very thanks!
Sorry my english, it is not very good!

jsalatas, 2019-09-23:
> You just have to integrate it in plasmoid.

Well.... if it seems to you too easy to do so, maybe you can just do it yourself and send a PR? :)
For me it's hard to do as I don't own that hardware. 

PS: No offense but I'm closing this now, as it is meaningless to continue discussing how easily it can be done.

mibofra, 2019-10-18:
I'm trying to give it support, as I'm interested in this feature too. Maybe, isw can be setted to be used as root marking the suid option for the binary, or simply asking a password to run it as root? And verifiying the presence of /etc/isw.conf , manually setup before. Probably, the CoolerBoost is the simpler option, maybe the hard one could be setting up the fan at a specific speed.

jsalatas, 2019-10-18:
> I'm trying to give it support, as I'm interested in this feature too.

Great! I'm reopening it then! 

>  Maybe, isw can be setted to be used as root marking the suid option for the binary, or simply asking a password to run it as root?

You don't need to that. The widget's set_prefs.sh script whixh is responsible for modifying the settings already runs as root (sudo) without the need of password.