jsalatas/plasma-pstate #36: Changing Frequencies not working

swissbuechi, 2019-10-02:
Hello

I use an Intel(R) Core(TM) i5-7200U CPU on my HP Elitebook 1040 x360 G2 with Kubuntu 18.04 installed.

The widget does not display my current Min/Max settings of the CPU/GPU frequencies and moving the fader does not work.

I found this link in your reddit comments: [](https://github.com/jsalatas/plasma-pstate/blob/master/99-plasma-pstate)

I tried executing it like:
`./set_prefs.sh -read-all
{"cpu_min_perf":"12","cpu_max_perf":"100","cpu_turbo":"true","gpu_min_freq":"300","gpu_max_freq":"1000","gpu_min_limit":"300","gpu_max_limit":"1000","gpu_boost_freq":"1000","gpu_cur_freq":"300","cpu_governor":"powersave","energy_perf":"balance_performance"}
`
But I dont really know further...
Hope you can help me.

Thank you in advance.

jsalatas, 2019-10-02:
> The widget does not display my current Min/Max settings of the CPU/GPU frequencies and moving the fader does not work.

I guess that the values are displayed but you just can't change these right? If this is not the case then I'm afraid that I don't know what the "The widget does not display my current Min/Max settings" and you may need to provide me a screenshot.

swissbuechi, 2019-10-02:
Sorry for my bad english.

https://imgur.com/a/J42etVg

jsalatas, 2019-10-02:
That's interesting! 
How did you installed it? Did you use the master branch or the 1.0.6 release?

swissbuechi, 2019-10-02:
Changing the CPU Govenor and Energy performance works correct.

jsalatas, 2019-10-02:
Did you use the master branch or the 1.0.6 release?

swissbuechi, 2019-10-02:
I think I used the master branch.

jsalatas, 2019-10-02:
There is an uninstall script. After you uninstalled it just download it from here 

https://github.com/jsalatas/plasma-pstate/releases/tag/1.0.6

unzip it an install it again like previously:

cd plasma-pstate
sudo ./install.sh

swissbuechi, 2019-10-02:
I did all the steps.
But now I get this error durring installation:

```
raphi@DKNB-RBU:~$ sudo ./Downloads/plasma-pstate-1.0.6/install.sh 
[sudo] password for raphi: 
cp: cannot stat '99-plasma-pstate': No such file or directory
Error: Installation of  failed: No such file: 
chmod: cannot access '/usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh': No such file or directory
chown: cannot access '/etc/sudoers.d/99-plasma-pstate': No such file or directory
chmod: cannot access '/etc/sudoers.d/99-plasma-pstate': No such file or directory

Setup complete.
```

jsalatas, 2019-10-02:
cd ./Downloads/plasma-pstate-1.0.6/
and then 
sudo ./install.sh

swissbuechi, 2019-10-02:
Sorry, my bad.

Now it installed with no errors:
```
raphi@DKNB-RBU:~/Downloads/plasma-pstate-1.0.6$ sudo ./install.sh
kf5.kpackage: Generated  "/usr/share/plasma/plasmoids//kpluginindex.json"  ( 140  plugins)
Successfully installed /home/raphi/Downloads/plasma-pstate-1.0.6/gr.ictpro.jsalatas.plasma.pstate

Setup complete.
```

Sadly the problem with the frequencies still exists.

jsalatas, 2019-10-02:
OK! Thanks for testing it! I need to check and come back to you later.

Could you please provide me the following details about your system?

1) distro you are using. 

2) plasma version you are using and if you installed it from the official distro's repos or from any other 3rd party repos (eg ppas). 

Thanks!

swissbuechi, 2019-10-02:
Thank you very much for your help.
I very enjoy your work!

Kubuntu 18.04

KDE Plasma Version: 5.12.9
KDE Frameworks Version: 5.44.0
Qt Version: 5.9.5
Kernel Version: 5.0.0-29-generic
OS Type: 64-bit

PPA: deb http://ch.archive.ubuntu.com/ubuntu/ bionic-backports main restricted universe multiverse
I added the bionic-backports PPA because of some KDE bugfixes in the later versions.

jsalatas, 2019-10-02:
I'll try to setup the same environment and let you know. 

It might take me some time. Thank you for your patience  :)

swissbuechi, 2019-10-02:
Thank you.
If you need some logfiles or a remote session on my machine, just let me know.

swissbuechi, 2019-10-23:
I recently updated to Kubuntu 19.10.
Now everything works fine.

jsalatas, 2019-10-23:
Thanks for the update! I guess I can close it now :\

closed 2019-10-23 by jsalatas

Rebitteration, 2020-01-03:
I can confirm this issue still exists. I have exactly the same problem on Kubuntu 18.04(.3) LTS release.
Maybe it'd be nice if you mention in your README.md that it does not work on this version of Kubuntu(/KDE 5.12.9).

cfitz25, 2020-04-01:
I also have this issue, if i restart the kernel it shows errors such as:
file:///usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/ui/main.qml:83: TypeError: Cannot read property 'unit' of undefined
file:///usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/ui/main.qml:74: TypeError: Cannot read property 'value' of undefined

My system is:
Kubuntu 18.04.4 LTS
KDE 5.12.9
Kernel 5.3.0-45-generic