jsalatas/plasma-pstate #38: Set number of online cpus

mibofra, 2019-10-18:
As in other tools for gnome, provided as extensions, it should be great to set the number of online cpus (using the chcpu command maybe?)

mibofra, 2019-10-18:
I'll try to see if in the same manner the plasmoid modify the cpu freq, I can set the number of the online cpus.

jsalatas, 2019-10-18:
> I'll try to see if in the same manner the plasmoid modify the cpu freq, I can set the number of the online cpus.

Thank you about this! I'm personally not interested in that feature, but if as a first step you could add support for it in the  set_prefs.sh script, it would be good to have!

mibofra, 2019-10-19:
I was trying to tweak the utils.js too and the qml ui to test it, but probably because I'm too tired or probably because you are faster with js, you could add it simpler than me to the ui.

Anyway, https://github.com/mibofra/plasma-pstate/blob/master/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh :D 

It reads the actual value from the selected cpu, and store it as true or false, as json, with the other values, in format cpu_NUM:state . It doesn't read cpu 0 state, as you can't set offline that cpu, in any case. On the GUI, it should be anyway visible, to be countable, with the other present cpus, maybe grayed out and not clickable, to outline that cpu 0 is present but can't be set offline.

On the cli, when you use a sudo ./set_prefs.sh -cpu-state NUM, toggle the switch of the selected cpu. So if the cpu was online, it is set offline. If offline, online.

For this reason, the plasmoid should, in my opinion, use switch widgets of each present cpu, and first display the actual cpu status, then issue the command to change the status. Maybe grouped not in column as for the other groups present in the widget, but in rows, one every 5 or 6 cpus/cores, to not take too much space.

At startup, all the cores, by default, are online. But, as the plasmoid should work in any case, that is why I wanted to know the state first, instead of declaring it already as in the case of the turbo.

If this seems a complication, we can simplify the code of the set_cpu_state function, to be of the same type of the set_cpu_turbo one. All the rest of the code added doesn't need changes in this case, as it's only there to enumerate the available cpus on the system.

As tested, the script works as described here, as expected :D

mibofra, 2019-10-22:
@jsalatas I've tried another simpler approach. Now I've added a slider to select the number of online cpus, and all in the code is set up to read and set the number of online cpus. I've tested it and it seems to work. I'm making a pull request.

P.S. I've noticed that if I set, as an example, 1, on the slider, for example:
`{'type': 'slider', 'text': 'CPUs', 'min': 1, 'max': 'cpu_total_available', 'sensor': 'cpu_online'},`
at every startup of the plasmoid, the first executed instruction is to change the selected sensor to the min value. I've set it to 0 and made in the script check for 0 as input, making it doing nothing.