jsalatas/plasma-pstate #40: New UI Proposal

jsalatas, 2019-10-20:
Hey guys!

With the various latest additions (and more to come in the near future), the widget seems to take too much screen space (see for example #39) and it is also not possible to add it to the system tray (see #7).
I though that it might be a good idea to redesign it in a more compact way. An idea is to present the various sections (Processor Settings, Energy Performance, Thermal Management, etc) in a way similar to the one used by the systray to present the hidden items and I tried to create a conceptual design which you can see at https://i.imgur.com/OvSpE8A.png  
 
What do you think about it? I would appreciate your feedback and also any additional ideas and designs! :)

Sparsa, 2019-10-21:
This looks amazing man, Keep up the good work :D

mibofra, 2019-10-25:
I agree, this is a nice mockup :)

arslancharyev31, 2019-10-30:
Looks great! Thanks for your efforts.

adojck, 2019-12-11:
Love it!

cmpute, 2020-01-06:
Definitely looks cool!

aancw, 2020-02-18:
Nice mockup. Can't wait to see!

brianmsm, 2020-04-24:
Amazing! I love it!

jsalatas, 2020-04-26:
Thanks guys! I am actually busy at some other tasks at the moment but I'll come back to this at some time soon (I hope) :)

rohitghali, 2020-05-13:
Super.

lapineige, 2020-09-17:
Very consistent with KDE Plasma design, that's very cool :)