jsalatas/plasma-pstate #44: Icon Tooltip works fine, but widget popup is empty

jpmgsxr, 2019-10-29:
Same issue from: https://github.com/jsalatas/plasma-pstate/issues/13

Difference is, only file in /etc/sudoers.d is 99-plasma-pstate.

If I "configure" and uncheck/recheck "use sudo for reading values" everything shows up but I cannot make changes.

jsalatas, 2019-10-29:
Didi you try the solution there? It should work as AFAIK opensuse also uses the wheel group instead of sudo

jpmgsxr, 2019-10-29:
I don't have wheel in there, just the 99-plasma-pstate for sudoers.d so his solution didn't work for me.

jsalatas, 2019-10-29:
What will happen if you run the following? 
`sudo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh -read-all`
does it ask you for a password?
what is the output?

jpmgsxr, 2019-10-29:
`sudo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh -read-all`

>{"cpu_min_perf":"11","cpu_max_perf":"100","cpu_turbo":"true","gpu_min_freq":"300","gpu_max_freq":"1100","gpu_min_limit":"300","gpu_max_limit":"1100","gpu_boost_freq":"1100","gpu_cur_freq":"467","cpu_governor":"powersave","energy_perf":"balance_performance"}

jsalatas, 2019-10-29:
does it ask you for a password?

jpmgsxr, 2019-10-29:
Yes

jsalatas, 2019-10-29:
It shouldn't ask you. 

What is the output of 
```
sudo cat /etc/sudoers.d/99-plasma-pstate
```

jpmgsxr, 2019-10-29:
sudo cat /etc/sudoers.d/99-plasma-pstate
[sudo] password for root: 
%wheel ALL=NOPASSWD: /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh

jpmgsxr, 2019-10-29:
https://github.com/jsalatas/plasma-pstate/issues/4 

Looking into this

jsalatas, 2019-10-29:
Doesn't make sense :\

Can you please restart your PC? :\

jpmgsxr, 2019-10-29:
Fixed. openSUSE doesn't add the user to the wheel group by default since it uses su. Added user to wheel and it is working fine. Sorry for the runaround.

GAlexMES, 2020-01-15:
@jsalatas I have the same issue with Ubuntu 19.10. Too answer your questions:

1. Does calling the script with sudo ask for a passwort: yes
2. What is the output of the script: The same as for jpmgsxr
3. Content of /etc/sudoers.d/99-plasma-pstate: The same as for jpmgsxr
4. Did I reboot: yes multiple times.

'groups wheel' returns 'wheel : wheel sudo'

Any ideas?

jsalatas, 2020-01-15:
> Content of /etc/sudoers.d/99-plasma-pstate: The same as for jpmgsxr

lol! That's an interesting one. I guess the installation script is confused with your username (wheel) as in some distros there is a wheel group instead of a sudo group. I guess I need to review it (maybe one can have a username called sudo). Anyway..... could you please modify the  /etc/sudoers.d/99-plasma-pstate and instead of 

```
%wheel` ALL=NOPASSWD: /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents
```
make it 
```
%sudo` ALL=NOPASSWD: /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents
```
then reboot and see if that fixed your issue? It guess you will be ok.

GAlexMES, 2020-01-16:
Content is now:
`%sudo ALL=NOPASSWD: /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh`

I rebooted. But still have an empty dialog.
Calling the script with sudo still needs a passwort.
Executing `id` gives me `... 27(sudo) ... `

jsalatas, 2020-01-16:
I honestly have no idea :\

Clearly the issue is that it still asks you for password when you run it, so you need to figure it out. I would recommend to uninstall the widget and then try installing again after you have removed the wheel user or something :\

Sorry that I'm unable to help you :(

GAlexMES, 2020-01-21:
I reinstalled and rebooted and still can't call it without a password.

> %sudo ALL=NOPASSWD: /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh

adojck, 2020-02-01:
> I reinstalled and rebooted and still can't call it without a password.
> 
> > %sudo ALL=NOPASSWD: /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh

Are you in `sudo` group? You can use `gpasswd -a <user> sudo` to add yourself to this group. Otherwise change `sudo` to `wheel` in `/etc/sudoers.d/99-plasma-pstate`

And note that your `/etc/sudoers.d/99-plasma-pstate` is not same like **jpmgsxr** has

DerAlSem, 2020-12-30:
I'd suggest to add a note to README.md for OpenSUSE users (or may be all users) to check, that they are in WHEEL group. As soon, as i've added my user to wheel group the popup started to show all it supposed to show.