jsalatas/plasma-pstate #46: Applet will spam "modprobe nvidia" if nvidia driver is nouveau

Joey11904, 2020-01-31:
The pstate applet appears to be trying to load the nvidia driver if nouveau is installed and in use

**System specs**
Operating System: Arch Linux 
KDE Plasma Version: 5.17.5
KDE Frameworks Version: 5.66.0
Qt Version: 5.14.1
Kernel Version: 5.3.13-zen1-1-zen
OS Type: 64-bit
Processors: 4 × Intel® Core™ i5-7200U CPU @ 2.50GHz
Memory: 11.5 GiB of RAM

**How to reproduce the issue:**
1) Install the nvidia drivers (I installed the `nvidia-dkms` package)
2) Compile and install this: https://aur.archlinux.org/packages/optimus-manager/
3) Using optimus-manager, set the default GPU to Intel GPU
4) Install the applet and enable it
5) Open htop and you will see `modprobe nvidia` running indefinitely

**The only way to prevent this issue from occuring is:**
1) Using optimus-manager, change the default GPU to nvidia (not ideal for running on battery)
2) Disable nvidia gpu from BIOS (some BIOSes might not support this

Extract the file to `/etc/optimus-manager/` then reboot for changes to take effect
[optimus-manager.conf.zip](https://github.com/jsalatas/plasma-pstate/files/4140591/optimus-manager.conf.zip)

If the above config file does not work try experimenting with the options presented here: https://github.com/Askannz/optimus-manager/wiki/A-guide--to-power-management-options

jsalatas, 2020-01-31:
It seems to me that the issue is the optimus-manager requires proprietary nvidia drivers to run. Plase have a look the the [README file](https://github.com/Askannz/optimus-manager#installation). It clearly mentions the following 

> Naturally, you must have the proprietary nvidia driver installed on your system. On Archlinux, you can use the packages nvidia or nvidia-dkms

Joey11904, 2020-02-01:
I already have the nvidia-dkms driver installed when installing the applet. What i meant is that the applet detects the nvidia driver and tries to modprobe it while the nouveau driver is in use.

Optimus-manager turns off the nvidia gpu (switches to nouveau) when the Intel GPU is in use. The applet tries to enable powermizer by modprobing the proprietary nvidia driver.

jsalatas, 2020-02-01:
If that's the case then I guess that the widget is incompatible with optimus-manager. If you could send a patch that fixes it then I would be happy to merge it.

jsalatas, 2020-02-01:
BTW: I just had a look at optimus-manager wiki at 
https://github.com/Askannz/optimus-manager/wiki/A-guide--to-power-management-options#configuration-2--pci-power-control 

in section "Configuration 2 : PCI power control" explains some issues with your configuration and proposes ways to fix these.

Joey11904, 2020-02-01:
I would love to contribute a patch; unfortunately, I don't know how to code.
And it appears that adding these lines:
`pci_power_control=yes`
`pci_remove=yes`
and deleting
`switching=nouveau`
solved the problem for me :)

(However I think this issue should be kept open since the solution above might not work for all nvidia laptops)

jsalatas, 2020-02-01:
I'm glad that your issue is fixed. Since closed issues are searchable, I guess I'll close it :)

closed 2020-02-01 by jsalatas

Joey11904, 2020-02-01:
UPDATE: Well technically it did solve the issue but now it's trying to run `nvidia-settings -q GpuPowerMizerMode` even when the nvidia gpu is removed from pci

Joey11904, 2020-02-01:
Could you add a condition that prevents the applet from querying powermizer if an nvidia gpu is not present?

jsalatas, 2020-02-01:
> Could you add a condition that prevents the applet from querying powermizer if an nvidia gpu is not present?

How would you do that? How you would detect in your system whether the nvidia gpu is not loaded? In any case I would need run a command to detect it. Right? So you would still have a command running. It would just not be nvidia-settings but something other. Will that make any difference to you?

Joey11904, 2020-02-01:
I guess you're right; adding extra code  would just complicate things more. 
I guess the best way for now would be to rename the nvidia-settings binary to something else to prevent the querying from happening.
The nvidia driver technically doesnt need the nvidia-settings GUI to operate properly so I guess this issue can be closed :)