jsalatas/plasma-pstate #47: Adding caching for heavy operations

Knebergish, 2020-03-26:
Hello.

Without using the plasma-pstate widget, my computer in idle mode consumes a very small amount of processor time: https://i.imgur.com/Vv7gjII.png

But if I turn on this widget with all the settings turned on (except for LG-specific ones), then it becomes obvious that it consumes a lot of processor time: https://i.imgur.com/03rkjhc.png
Every two seconds it is executed set_prefs.sh, which significantly loads the processor.

After a little research, it turned out that the main load is given by the checks "check_nvidia" and "check_dell_thermal", as well as setting the values ​​of the variables "thermal_mode" and "powermizer" (this is all in set_prefs.sh).
It seemed too expensive to perform these operations every two seconds, so I added writing new values of these settings to the file in the "set_thermal_mode" and "set_powermizer" procedures, and reading them from this file in "read_all". The execution of "check_nvidia" and "check_dell_thermal" has been removed.
As a result, the CPU load is significantly reduced and becomes indistinguishable from the load without using the widget: https://i.imgur.com/vfGBoSi.png
All settings remained functional and correctly changed their values: https://i.imgur.com/fRRzbVj.png

In my opinion, adding such caching will reduce unnecessary CPU load. It is unlikely that these settings will be changed without using the widget, but for this situation, we can provide a cache reset button (or periodically reset them, but less often than once every two seconds).

jsalatas, 2020-03-26:
Well, actually the widget is just a hack: a gui that calls shell scripts. And there are different attempts by different people to counteract many of its inefficiencies. mostly were oriented to the number of messages in the syslog, but it's the same case actually. 

Anyway.. I'm not sure I want to introduce a "cache feature". Sorry :(

But you can for sure hack it for your personal needs :)

userdeveloper98, 2020-04-06:
Hey @Knebergish could you please share you changes ?
I noticed the same problem
No widget:
[image]
With widget:
[image]

jsalatas, 2020-04-07:
I would also be interested on having a look at that code. :)

Knebergish, 2020-04-09:
This code is as simple as possible - I write the value of settings to a file when they are changed, and read them from this file in "read_all" function.

dadd390b5c24c1f826954b816ed66cdccbec6bd6

closed 2020-05-21 by Knebergish