jsalatas/plasma-pstate #48: [Manjaro KDE] Missing Dell Thermal Management feature after system update

zburgermeiszter, 2020-03-29:
Hi,

I'm using Manjaro linux, and until recently this Plasma widget has worked properly but after an update the Thermal Management feature has gone missing.

Do you have any idea what should I check or reinstall?

AFAIK it relies on the `libsmbios` package which I already removed and reinstalled without any success to get the Thermal Management back.

Thanks.

[image]

[image]

[image]

jsalatas, 2020-03-29:
> AFAIK it relies on the `libsmbios` package which I already removed and reinstalled without any success to get the Thermal Management back.

Do you still have `smbios-thermal-ctl` executable installed?

zburgermeiszter, 2020-03-29:
```
➜  ~ which smbios-thermal-ctl 
/usr/bin/smbios-thermal-ctl

```

jsalatas, 2020-03-29:
can you run the following and provide me the output?

`sudo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh -read-all`

it shouldn't ask you for a sudo password

zburgermeiszter, 2020-03-29:
That's correct, it did not ask for sudo password.

The output is:
```
{
  "cpu_min_perf": "9",
  "cpu_max_perf": "100",
  "cpu_turbo": "true",
  "gpu_min_freq": "300",
  "gpu_max_freq": "1150",
  "gpu_min_limit": "300",
  "gpu_max_limit": "1150",
  "gpu_boost_freq": "1150",
  "gpu_cur_freq": "1017",
  "cpu_governor": "powersave",
  "energy_perf": "balance_performance",
  "thermal_mode": "cool-bottom"
}
```

I just found the issue #3 , which mentions an issue with `$LC_ALL`. Apparently that is not configured properly.
When I open a new terminal I get the following error:
```
manpath: can't set the locale; make sure $LC_* and $LANG are correct
```
Does it affect the pstate plugin's ability to manage thermals?

jsalatas, 2020-03-29:
can you try the proposed solution?

https://github.com/jsalatas/plasma-pstate/issues/3#issuecomment-451703114

it is probably the same issue (locale related)

zburgermeiszter, 2020-03-29:
Fixing the `$LC_ALL` issue automatically fixed the pstate widget.

Instead of hacking the script, I fixed the `$LC_ALL` globally.
Here is the solution: https://forum.manjaro.org/t/default-locale-not-set-properly/64163/28

`sudo echo $LC_ALL` is still empty, but I can't see the locale error anymore in my terminal and the plugin is working properly again.

Thanks for your help.

Closing the issue now.

closed 2020-03-29 by zburgermeiszter