jsalatas/plasma-pstate #50: Option for a mote informative icon and/or susbtitute the icon with the temperature?

kupiqu, 2020-04-17:
I like that the tooltip informs about cpu use, fan revolutions and temperature. I found myself checking it every now and then to get that information.

Currently the icon is nice but not informative. I wonder if you would consider adding options such as:

- chage the regular color of the icon to orange first and then to red when the system temperature surpasses two increasing temperature thresholds

- an alternative that may like to some is replacing the icon by the system temperature

jsalatas, 2020-04-17:
I'm not doing it because I'm bored. I would gladly accept any PR though :)

kupiqu, 2020-04-17:
Fair enough. Running busy right now, but I may come back at some point. Others please jump in if you think it's a good idea...