jsalatas/plasma-pstate #54: Configuration not not saving, always reset

agungprabawa, 2020-05-31:
Hi, I got problems about saving the config, so after turn on or after wake up the PC the configuration become default again -> max 100% cpu usage and etc.
How can I save my configuration ?, thanks

jsalatas, 2020-05-31:
IMHO you should use some other power management tool (like TLP, powertop, etc) to do so. The widget is supposed to be just a GUI shortcut to easily override (temporarily) the settings of the underlying  power management tool.

Patana98, 2020-06-18:
Also I would really need it, it has no real use for me if for every start I have to set everything up again.
Is there a way I can save my configuration?

Patana98, 2020-06-18:
Solved, thank you very much, very useful