jsalatas/plasma-pstate #55: Unable to show intel GPU frequency

QuakeString, 2020-06-25:
I've installed this plasma from AUR.
it does not show intel GPU frequency. Please see attachment.
I've no idea which system detail I should give here. Please let me know, I'll provide

I'm running Manjaro KDE latest stable
[image]

jsalatas, 2020-06-25:
I see it detects Nvidia card. So probably the intel card is disabled or something? :\

EvgeniyMedyankin, 2020-07-19:
> I see it detects Nvidia card. So probably the intel card is disabled or something? :\

Didn't want to create a new request. 
**Could you tell me how to change the icon on the taskbar**.  
I want to install my, because this one does not display well. Thx