jsalatas/plasma-pstate #56: Widget seems to set anything...

jochen-01, 2020-07-13:
...it falls back to default.
As I understand the purpose of the widget (and the script) it should give the opportunity to temporary overwrite whatever is set system-wide.
If I click on, lets say, CPU Governor "Performance" it falls back to "Powersave" after a second or so.

Whats going wrong here?

Intel Core i5 10210U

jsalatas, 2020-07-13:
I have no idea! :)

it seems that you may either have some other software resseting it, or it just doesn't work :\

What distro you have?

jochen-01, 2020-07-13:
It's Ubuntu 18.04. I disabled tlp to avoid conficts. Therefore I dont't know of any other service resetting it. 
The energy configuration "tool" of Plasma 5 (I'm using KDE) does not deserve that name. As far as I see there is no option to configure clock and governor.

jsalatas, 2020-07-13:
> It's Ubuntu 18.04. I disabled tlp to avoid conficts.

it should work with no issues over TLP. You have sudo priviledges right?

Could you please run the following and let me know the output you get?
`
sudo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh
`

> The energy configuration "tool" of Plasma 5 (I'm using KDE) does not deserve that name. 

FYI: I'm working on it. :)
https://invent.kde.org/plasma/powerdevil/-/merge_requests/14

jochen-01, 2020-07-13:
Yes, I'm doing that as root. The installation script added 99-plasma-pstate to /etc/sudoers.d:
%sudo ALL=NOPASSWD: /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh

Should be o.k.

calling the script as you suggested gives:
Usage:
1: set_prefs.sh [ -cpu-min-perf |
                  -cpu-max-perf |
                  -cpu-turbo |
                  -gpu-min-freq |
                  -gpu-max-freq |
                  -gpu-boost-freq |
                  -cpu-governor |
                  -energy-perf |
                  -thermal-mode |
                  -lg-battery-charge-limit |
                  -lg-fan-mode |
                  -lg-usb-charge |
                  -powermizer ] value
2: set_prefs.sh -read-all

The actual values are (-read-all):
udo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh -read-all
{"cpu_min_perf":"9","cpu_max_perf":"100","cpu_turbo":"true","gpu_min_freq":"300","gpu_max_freq":"1100","gpu_min_limit":"300","gpu_max_limit":"1100","gpu_boost_freq":"1100","gpu_cur_freq":"300","cpu_governor":"powersave","energy_perf":"balance_performance"}

jsalatas, 2020-07-13:
could you please try the following?

`sudo /usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh -cpu-governor "performance"`
wait some seconds and then run 

`cat /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor`

The output (depending on the number of cpu cores) would be like the following
`
performance
performance
performance
performance
performance
performance
performance
performance
.....
`

if it says powersave instead of performance then try the following

`echo "performance" | sudo tee -a /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor`

and then again after waiting some seconds, run

`cat /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor`

jochen-01, 2020-07-14:
Always shows "powersave" for all cores.
cat /sys/devices/system/cpu/cpu*/cpufreq/scaling_available_governors
gives "performance powersave" for all cores. "performance" should be available...

jsalatas, 2020-07-14:
Probably you have some other software/setting that does this.

jochen-01, 2020-07-14:
O.k., I give up. It might be a tool of TUXEDO, who sold the device. Thank you for the effort.

jochen-01, 2020-07-14:
To complete it: It was really a tool from TUXEDO that did a reset on any change and replaced it with whatever has been configured by TUXEDO Control Center. This hint should help others. Thank you for your help.

jochen-01, 2020-07-15:
Hi,
meanwhile I did a few small changes in set_prefs.sh script to get around this issue. Values are set as usual and are reset with click on "Energy Performance" "default". The script stops configurable service(s) if it gets at least 2 parameters (i.e. changes are desired). The services are started with parameter -energy-perf default.

diff is:
```
2a3,7
> # Ralf <dlfkcf@gmx.net> 2020-07-15:
> # Disable specified services if values are changed.
> # That prevents unplanned resets of user changes.
> SERVICES="tccd"
> 
104a110,115
>     # Ralf, 2020-07-15: Start stopped service, if "default" has been selected
>     if [ $1 == "default" ]; then
>           for serv in $SERVICES; do
>                   systemctl start $serv
>           done
>     fi
245a257,266
> 
> # If script has been called with at least2 parameters, it should change one of
> # the values. Then disable all services ($SERVICES) to keep the change from
> # resets.
> if [ $# -gt 1 ]; then
>       for serv in $SERVICES; do
>               systemctl stop $serv
>       done
> fi
> 
```
Keep it if you decide to.

jsalatas, 2020-07-15:
oh! Sorry. I don't want to do it :)

As a user I would hate if a stupid widget messed up with the system's services. I hope this makes sense to you.

jochen-01, 2020-07-15:
Sure, no problem. I'll keep it an my laptop where it works for me. It would be better TUXEDO would fix it.
Funny that it was a tweet of TUXEDO leading me to your widget. A second issue I solved here is syslog flooting by your widget. But there has been another discussion where you made clear that you don't want to change it. It has been proposed to solve it by changing the code. My solution is pure sudoers.

Thank you for your work.

closed 2020-07-15 by jochen-01

jsalatas, 2020-07-15:
> A second issue I solved here is syslog flooting by your widget. 

I guess I'll fix it in the next major release of the widget. I'm currently working on implementing parts of what the widget does in kde's powerdevil which uses KAuth framework to get root priviledges, so I guess I'll eventually rewrite my midget to use KAuth as well. 
'

jochen-01, 2020-07-15:
Great news. Unfortunately it will take until one of the next LTS releases of Ubuntu until I'll see your work in KDE energy settings. ;-)