jsalatas/plasma-pstate #57: Doesn't show neither CPU not GPU frequency

rokuoku, 2020-08-02:
It was showing CPU frequency after installation, but at some point it disappeared (after reboot, I guess). GPU freq was never displayed.
[image]

jsalatas, 2020-08-02:
> GPU freq was never displayed.

I see it correctly6 detects nvidia. So probably the internal gpu is disabled, or something like that. 

I'm not sure about the frequencies thought. Which distro are you using? What plasma version? Also which laptop is that? I guess it is a Dell, but whhixh model?

rokuoku, 2020-08-02:
It is Dell Inspiron 7590, Manjaro, Plasma 5.19.3
I use prime offloading, so iGPU is not disabled , but it seems that xorg is running on nvidia. I will try to change that.

jsalatas, 2020-08-02:
> It is Dell Inspiron 7590, Manjaro, Plasma 5.19.3

Should work. Maybe some weird device naming or somrething :\

grigorii-horos, 2021-02-18:
@jsalatas I have same issue

jsalatas, 2021-02-18:
@grigorii-horos probably you have a dedicated GPU or something? :\

In any case, I'm not maintaining the widget anymore, however the great KDE team is working on having hardware power profiles directly in KDE, we just need to be patient :)