jsalatas/plasma-pstate #59: Automatic detection of nvidia-settings prevents dGPU from powering off

giddie, 2020-09-01:
My Dell Inspiron 15 7590 laptop contains an Nvidia GTX 1650 dGPU that is capable of dynamic power management:

https://download.nvidia.com/XFree86/Linux-x86_64/435.17/README/dynamicpowermanagement.html

However, when the `plasma-pstate` widget detects that `nvidia-settings` is installed, it appears to hold some lock open or poll the dGPU, preventing it from completely switching off.

It would be useful to have easy access to the power level settings from this widget, but if there's no way to prevent the issue, could we have an option to turn off detection of `nvidia-settings`?

jsalatas, 2020-09-01:
Um... I'm currently in the middle of implementing some of the widget's functionality in the core kde plasma (see also https://github.com/jsalatas/plasma-pstate/issues/58) and after that I'm planning to reimplement the whole widget from scratch. So please give me some time and we will probably handle it correctly. 

In the mean time as a workaround I guess you can modify the gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh and make it not to check for nvidia presence.

giddie, 2020-09-03:
Awesome; looking forward to seeing the fruits of your labour :)

For anyone else running into this: I've worked around the issue by commenting out the body of the `check_nvidia` function in the `set_prefs.sh` file @jsalatas mentioned, and just returning 1:

```bash
check_nvidia () {
    return 1
    # nvidia-settings -q GpuPowerMizerMode > /dev/null 2>&1
    # OUT=$?
    # if [ $OUT -eq 0 ]; then
    #     return 0
    # else
    #     return 1
    # fi
}
```