jsalatas/plasma-pstate #61: Plasma crash

openmindead, 2020-09-13:
Hi,
After updating Qt to 5.15.1 this widget causes Plasma to crash.
Just FYI.
Some logs:
```
kf.plasma.quick: Applet preload policy set to 1
Connecting to deprecated signal QDBusConnectionInterface::serviceOwnerChanged(QString,QString,QString)
file:///usr/share/plasma/shells/org.kde.plasma.desktop/contents/views/Desktop.qml:38:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
Cyclic dependency detected between "file:///usr/lib/qt/qml/org/kde/kirigami.2/styles/org.kde.desktop.plasma/Units.qml" and "file:///usr/lib/qt/qml/org/kde/kirigami.2/styles/org.kde.desktop.plasma/Units.qml"
file:///usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/ui/main.qml:192:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/ui/FullRepresentation.qml:56:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/ui/FullRepresentation.qml:48:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/plasmoids/org.kde.desktopcontainment/contents/ui/main.qml:274:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/packages/org.kde.desktoptoolbox/contents/ui/ToolBoxRoot.qml:38:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/shells/org.kde.plasma.desktop/contents/views/Desktop.qml:188:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
trying to show an empty dialog
file:///usr/share/plasma/shells/org.kde.plasma.desktop/contents/views/Desktop.qml:146:19: QML Loader: Binding loop detected for property "height"
file:///usr/share/plasma/shells/org.kde.plasma.desktop/contents/views/Desktop.qml:146:19: QML Loader: Binding loop detected for property "height"
file:///usr/share/plasma/shells/org.kde.plasma.desktop/contents/views/Desktop.qml:38:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/plasmoids/org.kde.desktopcontainment/contents/ui/FolderView.qml:1092:17: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/plasmoids/org.kde.desktopcontainment/contents/ui/FolderView.qml:1084:17: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/plasmoids/org.kde.desktopcontainment/contents/ui/FolderView.qml:1076:17: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/plasmoids/org.kde.desktopcontainment/contents/ui/FolderView.qml:1068:17: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/lib/qt/qml/QtQuick/Controls/Styles/Plasma/ScrollViewStyle.qml:60:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/plasmoids/org.kde.desktopcontainment/contents/ui/FolderView.qml:209:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/plasmoids/org.kde.desktopcontainment/contents/ui/FolderView.qml:170:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/plasmoids/org.kde.desktopcontainment/contents/ui/FolderView.qml:161:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/plasmoids/org.kde.desktopcontainment/contents/ui/FolderViewLayer.qml:191:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/plasmoids/org.kde.desktopcontainment/contents/ui/FolderViewLayer.qml:167:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/plasmoids/org.kde.desktopcontainment/contents/ui/main.qml:344:13: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/plasmoids/org.kde.desktopcontainment/contents/ui/main.qml:274:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/packages/org.kde.desktoptoolbox/contents/ui/ToolBoxRoot.qml:38:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/shells/org.kde.plasma.desktop/contents/views/Desktop.qml:188:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
trying to show an empty dialog
file:///usr/share/plasma/shells/org.kde.plasma.desktop/contents/views/Desktop.qml:146:19: QML Loader: Binding loop detected for property "height"
file:///usr/share/plasma/shells/org.kde.plasma.desktop/contents/views/Desktop.qml:146:19: QML Loader: Binding loop detected for property "height"
file:///usr/share/plasma/wallpapers/org.kde.image/contents/ui/main.qml:76:9: Unable to assign [undefined] to QStringList
file:///usr/share/plasma/wallpapers/org.kde.image/contents/ui/main.qml:75:9: Unable to assign [undefined] to int
Error: cannot change the containment to AppletsLayout
Error: cannot change the containment to AppletsLayout
file:///usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/ui/Slider.qml:68:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/ui/Slider.qml:68:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/ui/Switch.qml:46:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/ui/Slider.qml:68:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/ui/Slider.qml:68:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/ui/Slider.qml:68:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[... cut ...]  function onFoo(<arguments>) { ... }
file:///usr/lib/qt/qml/org/kde/plasma/components.3/ComboBox.qml:117:9: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
file:///usr/share/plasma/plasmoids/gr.ictpro.jsalatas.plasma.pstate/contents/ui/ComboBox.qml:50:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
```

Thiago4532, 2020-09-13:
I am having the same trouble since the last update.

jsalatas, 2020-09-14:
Um.... It works OK with 5.15.0 in kde neon. :\

Which distro is this? Please tell me this is not arch or arch based? :)

Thiago4532, 2020-09-14:
I literally use Arch Linux. The problem is the new version of Qt 5.15.1, I solved it by downgrading Qt to 5.15.0.

Thiago4532, 2020-09-14:
Qt 5.15.1 was released to fix some bugs. Maybe it's Qt fault, not yours.

jsalatas, 2020-09-14:
Hmmmmm..... I doubt that qt 5.15.1 introduced issues enough to break a widget that works OK with 5.15.0. let's leave it open until 5.15.1 comes to kde neon, as I don't intent to setup an arch distro to troubleshoot :)

jsalatas, 2020-09-14:
BTW: please have a look at this comment and the whole thread in general. Seems like an arch issue (again) to me :(

https://forum.manjaro.org/t/unstable-update-2020-09-13-kernels-qt-5-15-1-browser-pamac/24791/14

I'm closing it now, but please reopen again if anything else is working ok and only this widget crashes.

Thanks!

closed 2020-09-14 by jsalatas

openmindead, 2020-09-15:
Yes, this is being experienced on an Arch-based distro, Manjaro. The comment you pointed out is mine.
Does Neon really receive Qt updates? I thought it doesn't. Well, you'll see this bug then, too.

Knebergish, 2020-10-02:
Yesterday I received a stable update of Manjaro with the new Qt 5.15.1. After rebooting, plasmashell crashes with Segmentation fault: 
```
systemd-coredump	Process 1765 (plasmashell) of user 1000 dumped core.
Stack trace of thread 1765:
#0  0x00007fcb6ceef758 n/a (libQt5Gui.so.5 + 0x49d758)
#1  0x00007fcb6cef30ce _ZNK17QGridLayoutEngine22ensureColumnAndRowDataEP18QGridLayoutRowDataP14QGridLayoutBoxPKdS5_N2Qt11OrientationEPK24QAbstractLayoutStyleInfo (libQt5Gui.so.5 + 0x4a10ce)
#2  0x00007fcb6cef34d6 _ZNK17QGridLayoutEngine8sizeHintEN2Qt8SizeHintERK6QSizeFPK24QAbstractLayoutStyleInfo (libQt5Gui.so.5 + 0x4a14d6)
...
```

Other people had the same problem because of the Eventcalendar widget: [Manjaro Forum](https://forum.manjaro.org/t/stable-update-2020-10-01-kernels-mesa-kde-framework-deepin-systemd-pamac-xorg-server-qt/29141/38)
They were helped by its removal. I had the same problem with widget plasma-pstate, and deleting it solved the problem.
People say that in version 69 of widget Eventcalendar the problem is fixed: [Crash on arch](https://github.com/Zren/plasma-applet-eventcalendar/issues/154#issuecomment-692678574), but I still don't know how.

At the moment, I don't know how to fix this except by rolling back the Qt version to an earlier one.

mdedetrich, 2020-10-02:
Having the same problem, although in my case its interesting because on one laptop its working and on another version its not. The only difference I can see between the laptops is the fact that one has an nvidia blob running with prime (this is the laptop with the issue) where as the other laptop is just intel with an integrated GPU.

@jsalatas This seems to be specifically a widget issue to me, at least for my system everything else is working and uninstalling the widget (on the problematic laptop with nvidia blob) fixed the issue for me.

Maybe something has changed in how the widget gets details about the graphics card, so it may be more of a kernel issue than a KDE issue (just speculating here)

openmindead, 2020-10-02:
Indeed, this is related to Nvidia. When I use Intel-only mode (optimus-manager user here), this widget works fine and doesn't crash Latte-dock (or Plasma shell - depend on where it is being added / used). Both Nvidia and Hybrid modes trigger crashes.
This is not related to nvidia modules though, I tried to modprobe them in Intel mode, they loaded fine, and re-launching Latte with them loaded was successful, no crashes. At this point my observations are over. No ideas what else could be done to investigate further.

Knebergish, 2020-10-03:
@openmindead You can try using [this](https://github.com/jsalatas/plasma-pstate/issues/63#issuecomment-703066561) solution to temporarily fix the problem until a better solution is found.

openmindead, 2020-10-03:
@Knebergish I don't use this widget anymore since I've found `auto-cpufreq` but thanks anyway Artyom.

Thiago4532, 2020-10-04:
> @Knebergish I don't use this widget anymore since I've found `auto-cpufreq` but thanks anyway Artyom.

I still prefer this widget 'cause it's really good to control Dell Thermal Management. I downgraded Qt to 5.15.0 until the next Qt update.

mdedetrich, 2020-10-06:
Created an upstream issue at QT 5.15.1 here https://bugreports.qt.io/browse/QTBUG-87189