jsalatas/plasma-pstate #63: plasma-pstate widget causing plasmashell to crash in latest manjaro stable

mdedetrich, 2020-10-02:
After applying the latest manjaro stable update (details can be seen [here](https://forum.manjaro.org/t/stable-update-2020-10-01-kernels-mesa-kde-framework-deepin-systemd-pamac-xorg-server-qt/29141)), my KDE session after login becomes black if plasma-pstate is installed.

Interestingly this only happens on one of my laptops, i.e. the one that has an NVidia driver with the blob i.e. `video-hybrid-intel-nvidia-440xx-prime`. 

I made a post on the manjaro forums at https://forum.manjaro.org/t/stable-update-2020-10-01-kernels-mesa-kde-framework-deepin-systemd-pamac-xorg-server-qt/29141/130?u=mdedetrich but also reporting it upstream here.

Note that if I uninstall the https://aur.archlinux.org/packages/plasma5-applets-plasma-pstate then everything works fine.

Knebergish, 2020-10-02:
Hello.
I have the same problem, already wrote here: [This widget is causing Plasma to crash after update to Qt 5.15.1 #61](https://github.com/jsalatas/plasma-pstate/issues/61)

jsalatas, 2020-10-02:
> Interestingly this only happens on one of my laptops, i.e. the one that has an NVidia driver with the blob 
> ....
> Note that it seems that this KDE update in general is having issues with certain plasma widgets (and this appears to be one of them).

So apparently it is an NVidia/Qt issue that needs to be fixed here? Right? And not a Qt/widget issue. I mean if **_everything_** works OK in machines not using NVidia and break only in machines with NVidia, someone would just try to fix NVidia and not **_everything_**, whatever that **_everything_** might be. 

Do we all agree with this? :)

mdedetrich, 2020-10-02:
Yeah, now that @Knebergish has confirmed that in his case its also when using the NVidia blob it is something to do with how Qt is interacting with NVidia.

Also confirming that NVidia itself works absolutely fine (kernel module is loaded etc etc), how exactly is the widget/QT interacting with NVidia?

jsalatas, 2020-10-02:
> how exactly is the widget/QT interacting with NVidia?

The question is meaningless but anyway, the widget just calls the nvidia-settings executable with some parameters, you can see the exact command in 
https://github.com/jsalatas/plasma-pstate/blob/master/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh#L40
https://github.com/jsalatas/plasma-pstate/blob/master/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh#L150
and also 
https://github.com/jsalatas/plasma-pstate/blob/master/gr.ictpro.jsalatas.plasma.pstate/contents/code/set_prefs.sh#L217

I don't know however how Qt in general and maybe Xorg interacts with the Nvidia driver.

jsalatas, 2020-10-02:
Could you also do me a favor and consider the possibility that it is a manjaro/arch packaging/compiling issue either in nvidia driver or in qt? please? or is this out of question? :)

jsalatas, 2020-10-02:
BTW: has anyone trie the latest nvidia driver (450.80.2) that was released a couple of days ago?

https://www.nvidia.com/Download/driverResults.aspx/164073/en-us

mdedetrich, 2020-10-03:
@jsalatas Thanks for the help, I will test this out tomorrow. Since the widget is just calling the `nvidia-settings` executable I dont see how anything can be wrong here. There are 2 possibilities

1. There is some regression/change in `nvidia-settings` which is causing data to be saved that plasma-pstate cannot handle because it is not expecting it
2. There is some weird interaction with QT+NVidia which is causing some plasma widgets to fail for unknown reasons

Not sure what it is, I had a look at the stack traces and they don't seem to be that helpful

jsalatas, 2020-10-03:
Thanks for your help! I would be highly surprised it Qt 5.15.1 introduced some regression in nvidia-settings but you never know. It would still surprise me nevertheless. In your case I would focus more in compiling/packaging issues in Qt 5.15.1.

Knebergish, 2020-10-03:
A little research has shown that NVIDIA is not to blame.
Everything was falling because of NVIDIA, because only NVIDIA used the visual component ComboBox.
It turned out that everything works great if you comment out this line: [ComboBox.qml](https://github.com/jsalatas/plasma-pstate/blob/679c14321c1496bf12ca0487c2a520f3361cc98a/gr.ictpro.jsalatas.plasma.pstate/contents/ui/ComboBox.qml#L84)
`Layout.minimumWidth: parent.width`
Yes, now this ComboBox has an incorrect width, but it works -_-.

mdedetrich, 2020-10-03:
@Knebergish Perfect, thanks for looking into it. Seems like a change in Qt has meant that `parent.width` is an invalid value (either its `null`/hegative/too large). Might also be an initialization issue.

ntropy83, 2020-10-05:
Comenting out the line lets the widget die with an error for me "failed to load: and then referencing the ComboBox.qml" Yet commenting it out doesnt crash plasmashell no more.

jsalatas, 2020-10-05:
> Comenting out the line lets the widget die with an error for me "failed to load: and then referencing the ComboBox.qml" Yet commenting it out doesnt crash plasmashell no more.

umm....strictly from a software engineering perspective a "bug" in a widget (actually in any app) shouldn't be able to crash your whole desktop.

I honestly believe that the bug is not in the widget but somewhere else. I understand however that when the issue is fixed somewhere else, my widget might stop working (not crash the whole desktop) and I might need to change it somehow. Until then however, I'm sure that there is nothing for me to do in the widget. 

Hope you understand the above :\

mdedetrich, 2020-10-05:
> Hope you understand the above :\

From a software engineers perspective either

1. That line of code was relying on unspecified/undefined behavior
2. There is a regression in QT 5.15.1

We should investigate which case it is

jsalatas, 2020-10-05:
OK! Unfortunately my distro is still in Qt 5.15.0 so there's nothing much I can do from my side. I can't even try in a VirtualBox (my widget doesn't work in VM). So I see two ways to proceed here:

1. Either you use gdb in your PC and try to figure out where the crash of your whole desktop happen.
2. You create a simple example of a widget that crashes the whole desktop and can be tested in a VM and I will take care the gdb part from there. 

Thank you for you help! :)

closed 2020-10-05 by jsalatas

mdedetrich, 2020-10-06:
Sure I will have a look at this tonight, I will clone the project and try and figure out what is going on. I will also check QT's bug tracker to see if anything else is going on.

Also to be clear, it is very stupid that the bug is crashing the entire desktop (plasma shell to be more clear) but that doesn't make it any less of a bug.

mdedetrich, 2020-10-06:
Created an upstream issue for QT 5.15.1 here https://bugreports.qt.io/browse/QTBUG-87189

jsalatas, 2020-10-06:
I would report it in kde and not Qt, but anyway...

hygorhernane, 2020-10-11:
I'm using Fedora and was getting this crash too.

Its working now that i saw that variable was to blame. Replaced it with 
" Layout.minimumWidth: units.gridUnit * 4 "

 from another line to test and now its not crashing anymore and even working again.

mdedetrich, 2020-10-11:
@hygorhernane Do you want to set up a PR for this? At least it will fix it temporarily (and we can further investigate what the root cause is, it might be some time before a Qt release).

jsalatas, 2020-10-11:
@mdedetrich please don't ask other people to do things. It's actually getting annoying. :(

Clearly it seems that there is an issue in either qt5 or kde plasma, so (once again) you (I'm talking specifically about @mdedetrich  here) need to attach a debugger to plasmashell, figure the root cause and then create a simple example widget that shows the issue. Preferably you (again I'm talking specifically about @mdedetrich here) could create a PR for either kde plasma or qt, but in any case just finding the root cause of the issue and showing it with a simple example would be great. Otherwise you (@mdedetrich) are just making unhelpful noise :(

mdedetrich, 2020-10-11:
@jsalatas I am not sure what problem you have, but the only reason I didn't want to make a PR with @hygorhernane suggestion is because I didn't want to take credit for someone elses work. its a one line PR that anyone can do in 5 minutes so I have no idea why you are making a big deal out of it. My intention was to at least make a quick fix the widget so its working for other people because judging by the fact that Qt hasn't responded to the ticket its probably going to take some time time to figure out the root cause (ontop of this if the problem is with Qt then all NVidia users will have to wait for a new Qt release which obviously will take some time).

Secondly complaining that I am being "annoying" because I am doing other peoples work is hilarious when I spent time having to make a ticket on Qt while you spent the entire time weaseling the issue out of plasma-pstate's responsibility claiming that the issue has nothing to plasma-pstate and instead is a problem with Xorg/Nvidia (which is obviously wrong if you spent  a few seconds looking at the stack trace its either a Qt issue or a plasma-pstate issue or both) and to top it off making derogatory remarks such as "from a software engineering perspective". Mind you I also already spent enough time diagnosing the root cause of the problem when I upgraded my Linux's KDE/Qt upgrade.

So yeah, I can spend time hooking up a debugger to figure out what is going on but frankly and understandably I am not really motivated right now, especially considering that most of the "noise" is coming from your handling of the ticket (which I haven't seen elsewhere).

I'm done here, great job in making people feel welcome when they spend their own time (and in my own case more time than was necessary) properly diagnosing issues :+1:

jsalatas, 2020-10-11:
Once again: Thank you for your help diagnosing the issue. No one will look at you ticket however. The correct approach is what I already described at least twice: attach a debbuger to plasmashell, find the root of the cause that crashed the whole desktop, and either fix it or create a simple widget that demonstrates the issue and then and only then create an issue in KDE plasma. 

Can you do it? If yes, I would higly appreciate it! if no, then you can clone the code, "fix it" in any way you see fit and just wait until someone knowlegable enough can fix it the correct way, But please don't ask other people to create PRs and pleast don't bother QT and KDE developers with incomplete bug reports. We aren't your slaves you know :(

jsalatas, 2020-10-11:
BTW: I just decided that I don't want to maintain this widget any more. So you are on your own. I will fix however anything that affects me and only me. There! do what ever you like. Probably you (the users of the widget) would want to fork it and find a way to maintain it. 

Hope you are happy now.

mdedetrich, 2020-10-11:
> But please don't ask other people to create PRs

The alternative of having a crashing desktop when using your widget seems like its better

> pleast don't bother QT and KDE developers with incomplete bug reports.

Firstly the bug report is not incomplete (you should have a look at the other bug reports on Qt) and secondly the whole point of a ticketing system is for people to make bug reports. If there is not enough information in the bug report then someone can say so and I can provide more information. This is standard open source development from a "software engineering perspective".

No one is asking you to do anything, you are free to ignore this ticket completely and not do anything about it. Its clear that you don't feel like maintaining it which is fine, but then be honest about it (update the README so its in maintenance status) and don't guilt people making bug reports, if you don't have time/patience for the project then just say so rather then calling people "annoying" and "creating noise".

> Hope you are happy now.

Whether I am happy or not is not relevant, I am getting the impression that you didn't/couldn't spend the effort fixing the issue and so the status quo hasn't changed.

I will likely submit @hygorhernane quick hotfix as a patch to at least the archlinux AUR package so it doesn't crash peoples desktop. As you said, people are free to fork it if they want to maintain it.

Knebergish, 2020-10-12:
I am very sad to see such disputes here. Obviously, everyone here does what they can and want. We can't ask anyone to do anything. Such recriminations will not make opensource better or developers happier.
It's sad that @jsalatas is completely retired. Anyway, thank you for all the work you've done. Without you, this widget wouldn't exist at all.